letscontrolit/ESPEasy #710: NTP issue

PsychoAndi, 2018-01-12:
The NTP-function seems to have a bug. I use my ESPs from external places via a DynDNS service. When NTP is enabled, i can not address my ESP, every time i try this, i get an "IP blocked: ip.ad.dr.ess."

With not NTP enabled, i can login to the ESP from external...

TD-er, 2018-01-12:
Hmm did you have a pre-existing configuration?
I recently added a function to the **Mega** branch to allow for blocking traffic from outside the network.
See Webserver.ino 
https://github.com/letscontrolit/ESPEasy/blob/f7b37306aa38864e315aeb00dfa6cf7004d5ff4b/src/WebServer.ino#L56

And ESPEasy.ino
https://github.com/letscontrolit/ESPEasy/blob/f7b37306aa38864e315aeb00dfa6cf7004d5ff4b/src/ESPEasy.ino#L408-L422

And please check what version you are using.

PsychoAndi, 2018-01-13:
Oh yes, sorry...i used the Mega branch! No problems with the 2.0 release from tonight...

TD-er, 2018-01-13:
And was there a pre existing configuration in the flash?

PsychoAndi, 2018-01-13:
Yes, i think so...the ESP is in use since october, updated from time to time with newer versions of ESPEasy.

TD-er, 2018-01-13:
Ok and the behaviour kept changing when changing the NTP setting?
That would indicate overlap in settings

PsychoAndi, 2018-01-13:
Yes, when i changed the NTP setting, i also switched the ability to login into the ESP

TD-er, 2018-01-13:
Ok that sounds serious.
I will definitely look into this.
It is a 2.1 bug, not 2.0

Grovkillen, 2018-01-15:
I will move this to 2.1.0 if you think it's that level of commitment to fix this.

TD-er, 2018-01-15:
I will have a look at it this evening and will then conclude what version it really applies to.

TD-er, 2018-01-15:
I really cannot explain why this has happened.
I added the IP block settings to the SecurityStruct.
This SecurityStruct is stored in a separate file in the Spiffs (security.dat)
https://github.com/letscontrolit/ESPEasy/blob/0833029c828f054669057be8c5618f1b395bdf72/src/Misc.ino#L808-L816

The only thing I can imagine is that there is an issue with storing/updating existing files in the SPIFFS
Perhaps SPIFFS does not support fragmented files and they were stored very tight next to eachother?

s0170071, 2018-01-28:
have either of you tried this since there are checksums for settings in place ? If the bug is within load/save of the settings, this should be detected. If not, we could go ahead with #727

TD-er, 2018-01-28:
I have not been able to reproduce this.

PsychoAndi, 2018-01-28:
Hmph...i also can't reproduce it again. When i change the NTP setting now, it has no effect. But login into the ESP is only possible, when i do it from the local network. The JSON page display works from outside.

TD-er, 2018-01-28:
@PsychoAndi Did you check the setting for the IP filtering?
Config => Client IP filtering
Then check Client IP block level.
The default setting is "Allow local subnet" 
If you set it to "Allow All", you should be able to reach it from outside your network.

If you change it, please have a look at the NTP setting also to see if that setting has changed when setting the "IP filtering" flag.

s0170071, 2018-01-28:
Well, only remaining explanation would then be that NTP successfully hacked you :-O
No, seriously, I would consider this closed for now....

TD-er, 2018-01-28:
@PsychoAndi But is it now (after changing the IP filtering) possible again to reach the web interface from outside your network?

PsychoAndi, 2018-01-28:
No, not possible. Only when "Allow All" is selected.

TD-er, 2018-01-28:
OK, that is how this filter should work.
You are connecting from outside your own subnet, so "allow subnet only" should block that.
For your setup, you have to set to "Allow All" to be able to connect to the web interface from outside your network.
The idea is that having the web interface open too much by default is a bad idea, since it is easy to abuse.
Maybe we should add some extra subnet to allow for VPN use.
Something like "Allow local subnet + extra subnet"

TD-er, 2018-01-29:
I guess this one can also be closed?

Grovkillen, 2018-01-29:
Yep

closed 2018-01-29 by Grovkillen