letscontrolit/ESPEasy #712: Switch is set to dimmer upon update of FW

Grovkillen, 2018-01-13:
I made a OTA and my two switch devices were set to dimmer instead of switch.

https://www.letscontrolit.com/forum/viewtopic.php?f=6&t=3933&p=22485#p22485

TD-er, 2018-01-13:
That's a known issue.
It is introduced when fixing the non response on the switch.
Newly added switches will act correct and when you set  and save it again, it will be fixed

Grovkillen, 2018-01-13:
Yes it works but not optimal :)

Grovkillen, 2018-01-13:
Maybe a documentation is the only thing missing if the issue is not doable code wise.

TD-er, 2018-01-13:
Well it is possible to change it. We have 8 bits to store the setting. So I can add a check to see if it is an invalid choice.
Normally I am all in favor of "settings transitions", warning users etc. But here the space is tight.

Well I just thought of a new feature.. will add it in a sec :)

Grovkillen, 2018-01-13:
Hero :)

psy0rz, 2018-01-23:
@TD-er  this is the official place to do "upgrades": https://github.com/letscontrolit/ESPEasy/blob/v2.0/src/Misc.ino#L518

martinus used to update the buildnumber on each release, however for us now the buildnumber is 20000 for 2.0 and 20100 for mega/next v2.1.

Altough i'm not happy with that as well..so this needs some more thinking/discussing. :)

The buildnumber is also used to show the ESPEasy version number of all the nodes in the network via the custom udp protocol. 

Maybe we need a 'config-version' number instead? But we already have 2 other defines for this: https://github.com/letscontrolit/ESPEasy/blob/v2.0/src/ESPEasy.ino#L156

We those are changed it triggers a full format/reset.

We wil need to review all these and come up with just 2 variables i think: Version number and config-version. If the config-version changes, the BuildFixes (rename to DoUpgrades()) should decide what happens: "Can we upgrade the config, or should we format?"

Grovkillen, 2018-02-28:
Fixed right?

closed 2018-03-16 by Grovkillen