letscontrolit/ESPEasy #713: ESP Framed Display crash

s0170071, 2018-01-13:
v2.0, core 2.3.0 this time.
enter three lines of text, speed: very slow, timeout0, delay 1. It is almost constantly scrolling. Now delete it.
 

```

INIT : Booting version: (custom)
INIT : Warm boot #2
FS   : Mounting...
FS   : Mount successful, used 76053 bytes of 957314
INIT : Free RAM:24496
INIT : I2C
INIT : SPI not enabled
WIFI : AP Mode disabled
WIFI : AP Mode ssid will be ESP_Easy_11 with address 192.168.4.1
WIFI : Connecting MNET attempt #1
WIFI : Connected! IP: 192.168.1.32 (ESP_Easy_11)
WD   : Uptime 0 ConnectFailures 0 FreeMem 22368
FILE : Saved config.dat
FILE : Saved config.dat
FILE : Saved security.dat
FILE : Saved config.dat
FILE : Saved config.dat
FILE : Saved config.dat
FILE : Saved security.dat
WD   : Uptime 1 ConnectFailures 0 FreeMem 20160
FILE : Saved config.dat
FILE : Saved config.dat
FILE : Saved config.dat
FILE : Saved security.dat
WD   : Uptime 1 ConnectFailures 0 FreeMem 20848
FILE : Saved config.dat
FILE : Saved config.dat
FILE : Saved security.dat

Panic /home/john/ArduinoPortable/arduino-1.8.5_ESP2.3.0/portable/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/core_esp8266_main.cpp:98 __yield

ctx: sys 
sp: 3fff38f0 end: 3fffffb0 offset: 01b0

>>>stack>>>
3fff3aa0:  3ffe9a40 00000000 000003e8 3fff2fc0  
3fff3ab0:  00000000 3fff6d44 00000000 4023cb95  
3fff3ac0:  3fff04e4 00000000 3fff08fc 4022372d  
3fff3ad0:  3fff04e4 0000001f 00000001 40229b2e  
3fff3ae0:  00000014 00000400 3fff2829 402362a4  
3fff3af0:  00000000 00000000 00000000 402362cc  
3fff3b00:  000000bc 0000003c 3fff3cd0 3fff3d0c  
3fff3b10:  00000001 3fff7954 00000000 00000000  
3fff3b20:  00000000 00000001 3fff3d0c 3fff3cd0  
3fff3b30:  3fff3d0c 00000001 3fff3eb0 4022a4aa  
3fff3b40:  3ffeabb2 402441e3 3ffee320 3fff5107  
3fff3b50:  00343332 00000000 00000000 00000000  
3fff3b60:  00000000 00000000 00000000 00000000  
3fff3b70:  32343332 32333433 00000000 00000000  
3fff3b80:  00000000 00000000 00000000 00000000  
3fff3b90:  32343332 34333233 00000000 00000000  
3fff3ba0:  00000000 00000000 00000000 00000000  
3fff3bb0:  00000000 00000000 00000000 00000000  
3fff3bc0:  00000000 00000000 00000000 00000000  
3fff3bd0:  00000000 00000000 00000000 00000000  
3fff3be0:  00000000 00000000 00000000 00000000  
3fff3bf0:  00000000 00000000 00000000 00000000  
3fff3c00:  00000000 00000000 00000000 00000000  
3fff3c10:  00000000 00000000 00000000 00000000  
3fff3c20:  00000000 00000000 00000000 00000000  
3fff3c30:  00000000 00000000 00000000 00000000  
3fff3c40:  00000000 00000000 00000000 00000000  
3fff3c50:  00000000 00000000 00000000 00000000  
3fff3c60:  00000000 00000000 00000000 00000000  
[... 737 lines cut ...]
3fff6a80:  04400446 00000000 00000000 00000000  
3fff6a90:  00000000 041f06a1 04430449 00000000  
3fff6aa0:  00000000 00000000 00000000 00000000  
3fff6ab0:  0446044e 32343332 34333233 20202000  
3fff6ac0:  20202020 20202020 00000000 00000000  
3fff6ad0:  00000000 0075252e 04490463 00000003  
3fff6ae0:  00000000 00000000 00000000 00000000  
3fff6af0:  00000000 00000000 00000000 00000000  
3fff6b00:  00000000 00000000 00000000 00000000  
3fff6b10:  00000000 00000000 00000000 00000000  
3fff6b20:  00000000 00000000 00000000 00000000  
3fff6b30:  00000000 000000⸮
 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09826c6d
~ld
⸮U

```
`

psy0rz, 2018-01-13:
Ahh so it reboots on removal? I think i know why. Its probably uninitialized in delaybackground while still in use. This is a core bug that might affect others. Thanks

TD-er, 2018-01-13:
Indeed, must use an internal state to disable it before removal.
Perhaps even with a delay in between (delay() calls yield())

psy0rz, 2018-01-13:
yeah this is a bit difficult...since some plugins like thatone use delayBackground(), which in turn will handle webrequests, its possible to call the plugin cleanup function at that point. which will cause crashes.

perhaps we should set a cleanup-flag somewhere and let the mainloop do the cleaning up or something?

also the cleanup function is new, before nothing got cleaned up and kept consuming resources.

s0170071, 2018-01-13:
I think not crashing is good enough. Why not just ignore delete while executing a plugin? You would have to push delete again and that's it. So each plugin needs an execution flag. 
On the other hand we could also introduce a kill flag which causes the plugin to get killed instead of being called by the scheduler.

TD-er, 2018-01-13:
Perhaps a multi-level state, with bits indicating the next step to be performed.
So it becomes a state + ToDo.

e.g. a byte where every bit position indicates a task and when set to 1 it has to be set to 0 first before the next task could be done.

Tasks could be something like:

1. Enable running
2. Disable running
3. Reset to default
4. Delete

This means task 1 should be performed before 2 can be done, etc.
When this byte is set to 0, all tasks are performed.
And since it would be a state/tasklist at runtime, its definition can be changed with newer firmware releases without strange effects.

psy0rz, 2018-01-23:
I'll just fix this with a "shouldReboot" flag which warns the user to reboot to fully "unload" the plugin.

This way it wont crash anymore and we'll also fix #718  (since we dont have to recreate the objects every time)

psy0rz, 2018-01-23:
ok i've added an tested a seemless reboot function (just set shouldReboot=true). will try to add that to this plugins and other soon. need to sleep first since its 5am here now. i'm actually later than the buildbot :)

psy0rz, 2018-01-23:
try adding and removing the SDS011 plugin as an example. (note: reboot wont work just after flashing, this is an ESP issue, reset it one time)

s0170071, 2018-01-23:
I'll give it a try this evening. 
Seems that if we can spot when exactly to reboot (e.g. a plugin was removed) , the "submit" button could also say "submit and reboot" and just do save, reboot and reload the current page. It will take a second or two longer to load, but...

psy0rz, 2018-01-23:
perhaps. or even make a autoreboot option if it seems to function very well.

s0170071, 2018-01-23:
Yeah. Auto-reboot. I bet a euro or two that someone reports that as an issue.
Wait. Cant we just re-direct the auto reboots we already have ? Just kidding ;-)
Do you know what the reboot time would be if unnecessary parts were skipped (testing crc and spiffs etc.). Some sort of fast reboot.
With the espressif code in the background it appears imposible to just restart the ESPEasys part of the software while maintaining the network connectivity, right ?

psy0rz, 2018-01-23:
the reboot is actually superfast, the most time is spend while connecting to the wifi..dont think we can change that a lot.

TD-er, 2018-01-23:
Well the wifi code is a bit slow if the first given ssid is out of reach.

But software reboot may not always work.
I've seen it crash the ESP sometimes. (remember the issue where I was looking into reboots when it appeared some computer in my network had still a cronjob to call reboot :) )
Sometimes the breadboard node on my desk kept showing 5:59 or 6:00 on the display (6am was reboot)

Only fix then is to press the reset button.

s0170071, 2018-01-23:
I experienced this as well. Is there a sample code that reliably reproduces the failing resets/reboots ?

TD-er, 2018-01-23:
cronjob on another server which will call the reboot url ;)
I guess you still have to make sure the plugins will no longer called from the main loop.
So make the reboot a task and check every 1/10th of a second if a reboot is needed.  (not looked at the code, so might already be implemented like this)

psy0rz, 2018-01-24:
so there where multiple issues. also the framed oled in mega uses a different library that needs a call to display->end() before deleting it.

everything seems fixed now. (webinterface will be unresponsive during scroll now, cant fix that)

closed 2018-01-24 by psy0rz

psy0rz, 2018-01-29:
Woops forgot to commit the last fix in v2.0. just committed it.

commit 2f32799b (2018-01-24): fix #718. also update display on save and removed graphical boot logo to conserve space. also fixes #713. (web interface will be non repsonsive during scrolling now, but there is no easy fix for that)