letscontrolit/ESPEasy #714: Build:	20000 - Mega

s0170071, 2018-01-13:
sysinfo.html shows version 

> Build: | 20000  - Mega

when using core 2.3.0 and ESPEasy v2.0 
Why Mega ?

Grovkillen, 2018-01-13:
Its the name of the new features branch.

Grovkillen, 2018-01-13:
I will together with other core members make up a more extensive explanation with all these different builds. Even I have a hard time keeping track of it 🙈

psy0rz, 2018-01-13:
Yeah we need to make that more clear:)

TD-er, 2018-01-31:
Do the core members already have made up their minds on proper naming to distinguish v2.0 and Mega branch?

Grovkillen, 2018-01-31:
Ok, this is my idea.

"Mega" is what we're currently call the branch that at some point will spawn 2.1.0. At that stage "Mega" will die and become a new working branch called "Giga" and aiming for 2.x.0/3.0.0

The current 2.0.0 branch is a spawn from the never existing working branch of "kilo".

So in the 2.0.0 GUI nothing about "Mega" should be mentioned.

Thoughts on that?

Grovkillen, 2018-01-31:
We'll just use the SI metric prefix system. When we reach "yotta" we just roll over to "yocto" and start to work our way through the list again.

TD-er, 2018-01-31:
Or we could use the international spelling alfabet.
After "kilo" we get "lima" ;)

s0170071, 2018-02-01:
I am not core but since I brought this up I dare to comment ;-)
Those branch names were a riddle to me at first. May I suggest to use more clearly speaking names along with a nerdy "aka". For example

v2.0 (aka Einstein, stable)
v2.1 (aka Curie, alpha)
v2.2 (aka Cochrane, development)

Other than that, I would like the build date as part of the build string, at least for the dev branch. So instead of "20000 - Mega" the build string could read "2018_02_01 Mega". If the compiler does not support a timestamp, I can include it into crc2.py, along with the exact time if you want.

Grovkillen, 2018-02-01:
That sounds really nice @s0170071 , I tried to make sense of the current names ;)

Regarding the time stamp, I've been told that the this is not possible with the current framework so please add that functionality to your py script.

s0170071, 2018-02-01:
I'll add the timestamp, as string and integer. Using the integer (e.g.20180201) we could store the date of the binary in the settings and instead of breaking everybodys config (https://github.com/letscontrolit/ESPEasy/pull/665) we could implement a temporary switch to decode "old" configurations and store them as new ones. Using the time stamp you could nail that point in time when the binary made a critical change.

TD-er, 2018-02-01:
You can very easily add defines to the build string for the compiler and those defines will be usable in the code.
So adding a timestamp is no big deal.

s0170071, 2018-02-01:
@TD-er do I have to set this build string manually for each compile or can it automatically include the build time ? (like eg. gcc somefile.cpp -incTimeStamp)

svmac, 2018-02-01:
You can use macros __DATE__ and __TIME__ for compile's date and time, example:

    reply += F("<TR><TD>Build date:<TD>");
    reply += F(__DATE__);
    reply += F(" - ");
    reply += F(__TIME__);

will produce:

Build date:     Jan 28 2018 - 12:05:55

TD-er, 2018-02-01:
Maybe those __DATE__ and __TIME__ are even better choices to display instead of Git version.
I have used the Git version label for a while on my own system to mark builds from different branches.
But the releases only use v2.0 or Mega.

@s0170071 You can run some shell scripting from within the PlatformIO to resolve the time. But @svmac has an even better solution, which is already present.

Grovkillen, 2018-02-01:
Does Travis have that as well?

s0170071, 2018-02-01:
Agreed. We should include the date in the 20000 Mega string - if it is decided what it is going to be... 
@svmac wins, and I don't use platformio anyway.

s0170071, 2018-02-01:
Should I exclude the build date from the binary CRC ? Otherwise it may happen that building the same source on two different dates / times produces different CRCs.

TD-er, 2018-02-01:
@s0170071 What do you mean?
How is the build date now included in the CRC?
The CRC should be computed on the sketch firmware as it is being flashed to the ESP device.
So I guess the only thing that has to be excluded is the value of the CRC itself.

s0170071, 2018-02-01:
Currently there is no build date included in the binary. What I mean by excluding the date is the following: if you build a binary today, it gets a distinct hash vaule. If you build the exact same source tomorrow... then what ?
Should it : 
a) get the same hash because it is an functional identical binary (you then exclude the date from the hash) 
b) is it ok for it to get a different hash, then you inlcude the date. 

Option a) would make binaries comparable because they are the same not matter when you build.  
Option b) is what you propose. I would then make no sense to display the hash in the web interface because if a user reports an issue the hash only contains sensible information if the binary was pre-built.

I would go for a). What remains to be seen is wether a travis build , a Arduino IDE build and a Platformio build indeed yield the same hash for the same source.

s0170071, 2018-02-01:
It turns out that my comment above is more of theoretical nature. The CRC resides in an unchecked section of the memory and so is the date string.
I will wrap up everything in a global struct so that it's available for whatever...

TD-er, 2018-02-01:
You will get different CRCs when you're using these __DATE__ and __TIME__ values in the source, then the CRC will always be different.

TD-er, 2018-02-01:
As long as FEATURE_ADC_VCC is not set to true, it cannot be used.
Use the source: https://github.com/letscontrolit/ESPEasy/search?utf8=%E2%9C%93&q=FEATURE_ADC_VCC&type=

psy0rz, 2018-02-01:
Regarding "mega": This is the name of the main development branch. Ideally this just should be renamed "master" (like the git standard convention).

Normally master will always exist and always be the development branch. From this we will fork branches like v2.1, v2.2 etcetera.

What we should NOT do is rename the development branch on every new release. This makes development and contributing harder and more confusing.

I'm not sure if its possible to actually rename mega to master without causing too much trouble. Perhaps we should yes...branch it to master and place a huge warning in the readme.MD that says "DONT USE MEGA ANYMORE".

psy0rz, 2018-02-01:
@s0170071 do you want our slack development channels as well? its here https://espeasy.slack.com

TD-er, 2018-02-02:
@psy0rz Branching strategy is another (although related) topic.
But beware! Discussions on branching strategies are prone to achieve status of religious beliefs :)

s0170071, 2018-02-02:
@TD-er don't worry. I inialized a struct with it. Its the same struct that now holds the checksums. You can fetch the time from there like i did for the sysinfo output. Its all in https://github.com/letscontrolit/ESPEasy/pull/811

Grovkillen, 2018-02-10:
I have now added this: https://www.letscontrolit.com/wiki/index.php/ESPEasy#Version_names.2C_numbers.2C_and_branches

closed 2018-02-10 by Grovkillen

s0170071, 2018-03-23:
@psy0rz  the slack site is by invitation only. Will you send one?

> @s0170071 do you want our slack development channels as well? its here https://espeasy.slack.com>

Grovkillen, 2018-03-23:
I can invite you if you give me your email address. You can give it to my in a pm on the forum.