letscontrolit/ESPEasy #715: Framed display shows 12 lines

s0170071, 2018-01-13:
although only 3 are configured.
core 2.3.0, ESPEasy 2.0
 sorry, just tried it again ;-)

psy0rz, 2018-01-13:
You mean it tries to scroll to all 12 lines?

s0170071, 2018-01-13:
Need to verify. But I am sure it shows the 12 dots at the bottom.

psy0rz, 2018-01-13:
Mine doesnt show the dots:)

s0170071, 2018-01-13:
Maybe it's due to some mixup in the config data? I do switch back and forth between versions...
I will have a look at crc-ing this...

TD-er, 2018-01-13:
In the 2.0 version it will indeed show 12/lines per screen dots.
In the mega branch it will only show a dot per  page with data

psy0rz, 2018-01-13:
is this because of the lib update? should we close the issue stating: fixed in mega?

TD-er, 2018-01-13:
Nope, those were my own changes/improvements.
And since I made these changes on top of the latest lib update, it was moved to the Mega branch.

psy0rz, 2018-01-15:
ah ok. well we close the issue, its solved in mega.

closed 2018-01-15 by psy0rz