letscontrolit/ESPEasy #716: frequent writes of config.dat

s0170071, 2018-01-13:
core 2.3.0, ESPEasy 2.0

when I add a new device, serial output shows
```
FILE : Saved config.dat
FILE : Saved config.dat
FILE : Saved security.dat

```
the moment I select a device, even before I submit. 
When I do press submit I get another
```
FILE : Saved config.dat
FILE : Saved config.dat
FILE : Saved config.dat
FILE : Saved security.dat

```
Isn't that too much for the flash ?

Grovkillen, 2018-01-13:
Not a problem since you will most likely configure the unit very much after the initial install and perhaps occasionally re-configure.
But for beginners looking around in the GUI this will lead to the maximum flash write count hit maximum. So preferably this constant config.dat dump needs to be looked upon.

psy0rz, 2018-01-13:
Yeah i want to change the moment of saving anyway.

TD-er, 2018-01-13:
Another feature could be to allow the settings to be active and save the settings to flash when it is working like expected. Then a crash would not lead to an unusable configuration.

Grovkillen, 2018-01-13:
That is a good idea.

psy0rz, 2018-01-13:
oh thats even better indeed.:) an "save" button. however this is partially problematic: some plugin settings are loaded EVERY time the plugin is called form the mainloop (to save memory).

Grovkillen, 2018-10-23:
@TD-er what should we do with this one? It's in the scoop of our future JSON settings idea which we already are discussing here: #1292

closed 2018-10-23 by Grovkillen