letscontrolit/ESPEasy #717: Useless code in OpenHab MQTT controllerplugin _C005.ino ??

jodur, 2018-01-13:
See the part of the original code below marked Bold:

This part **makes no sense** when you send a generic MQTT message!

For example if you send a non CMD topic:

<sysname>/<taskname>/Anything <value> , the MQTT part is handled by the code marked in bold.

ESPEASY subscribes to: "<sysname>/#", so the value will be handled by CPLUGIN_PROTOCOL_RECV switch statement.

this results that plugins are called with   (PluginCall(PLUGIN_WRITE, &TempEvent, cmd));

cmd=<taskname>
TempEvent.Par1=<taskname>.toint() => results in zero because <taskname> is a string!!!
TempEvent.Par2=event->String2.toFloat(); =>original payload mqtt <value> is converted to float??
      
in my opinion the code should be as below, to handle generic published MQTT messages to ESPEASY

cmd="";  // CMD is empty so it is not a CMD (COMMAND)
TempEvent.String1=event->String1;
TempEvent.String2=event->String2;
TempEvent.Source = VALUE_SOURCE_MQTT; 

In this case the plugin would receive the original topic and payload in the PLUGIN_WRITE and general MQTT topic could be send and handled within the PLUGIN_WRITE.

The orginal part concerning CMD and EVENT will still working after these changes.

Is this a Bug or a design purpose i don't understand?

**Orginal code part of  _C005.ino:**

case CPLUGIN_PROTOCOL_RECV:
      {
        // Split topic into array
        String tmpTopic = event->String1.substring(1);
        String topicSplit[10];
        int SlashIndex = tmpTopic.indexOf('/');
        byte count = 0;
        while (SlashIndex > 0 && count < 10 - 1)
        {
          topicSplit[count] = tmpTopic.substring(0, SlashIndex);
          tmpTopic = tmpTopic.substring(SlashIndex + 1);
          SlashIndex = tmpTopic.indexOf('/');
          count++;
        }
        topicSplit[count] = tmpTopic;

        String cmd = "";
        struct EventStruct TempEvent;

        if (topicSplit[count] == F("cmd"))
        {
          cmd = event->String2;
          parseCommandString(&TempEvent, cmd);
          TempEvent.Source = VALUE_SOURCE_MQTT;
        }
        **_else
        {
          cmd = topicSplit[count - 1];
          TempEvent.Par1 = topicSplit[count].toInt();
          TempEvent.Par2 = event->String2.toFloat();
          TempEvent.Par3 = 0;_**
        }
        // in case of event, store to buffer and return...
        String command = parseString(cmd, 1);
        if (command == F("event"))
          eventBuffer = cmd.substring(6);
        else if
          (PluginCall(PLUGIN_WRITE, &TempEvent, cmd));
        else
          remoteConfig(&TempEvent, cmd);

        break;
      }

psy0rz, 2018-01-13:
Me neither. Will look into it soon since i will add it to the testsuit anyway.

TD-er, 2019-10-27:
This plugin has been changed a lot since this issue. Close for now.

closed 2019-10-27 by TD-er