letscontrolit/ESPEasy #718: Framed display loses 2k ram on each submit.

s0170071, 2018-01-13:
branch mega, core 2.4.0 and 2.3.0

- The ESP starts of with 21k ram and no devices. 
- When I add a "normal" display, there is almost no additional ram usage. 
- If I use the framed display, the ram drops to 19k after submit. 

If I submit again, it drops to 17k. I got as low as 6k until it became unresponsive.

TD-er, 2018-01-13:
In only the Mega branch, or also the v 2.0 branch?

s0170071, 2018-01-14:
2.0 seems unaffected.

TD-er, 2018-01-14:
Ok good to know.
I will look into it, to see if all new and delete are done correctly.
If I remember correctly I already added a virtual declaration to the destructor of the base class, but maybe I overlooked some?

TD-er, 2018-01-16:
Somewhat related: #354 ???

psy0rz, 2018-01-16:
yeah i think that is already done in 2.1  (its not called PLUGIN_STOP)

psy0rz, 2018-01-16:
yeah, i closed that one. it is called PLUGIN_EXIT. however the framed plugin is not yet using it.

Since this plugin is calling backgroundtasks() while scrolling, it would crash if one would delete the plugin while still scrolling.

If we dont call background task, the web-gui will be unresponsive during scrolling.

So we need some better solution for it. Perhaps set an unload flag for a task that needs unloading? Although that would have other implications as well...

psy0rz, 2018-01-24:
so there where multiple issues. also the framed oled in mega uses a different library that needs a call to display->end() before deleting it.

everything seems fixed now. (webinterface will be unresponsive during scroll now, cant fix that)

commit 2f32799b (2018-01-24): fix #718. also update display on save and removed graphical boot logo to conserve space. also fixes #713. (web interface will be non repsonsive during scrolling now, but there is no easy fix for that)
closed 2018-01-24 by psy0rz