letscontrolit/ESPEasy #719: DS18b20 incorrect reading after cold start

Oehoeboeroe, 2018-01-13:
### Steps to reproduce
-How can we trigger this problem?
NodeMCU+DS18B20 on GPIO2 (D4) , pullup 4k7
installed mega-20180104 (binary)
setup network settings
define Environent sensor DS18B20 on GPIO2

All readings are OK, expect after a powerup (cold start) the temp shows 85 degrees C.
After that all readings are fine...
WIFI : Connected! IP: 192.168.0.203 (ESP_Easy_0)
WD   : Uptime 0 ConnectFailures 0 FreeMem 20376
DS   : Temperature: 85.00 (28-d6-a5-59-5-0-0-f7)
WD   : Uptime 1 ConnectFailures 0 FreeMem 20376
WD   : Uptime 1 ConnectFailures 0 FreeMem 20376
DS   : Temperature: 21.50 (28-d6-a5-59-5-0-0-f7)

-Does the problem presist after powering off and on? (just resetting isnt enough sometimes)
Problem only after powerup, first reading is wrong, others are correct.
When device is resetted, first reading is OK (and others)

### Expected behavior
Tell us what should happen?
right temperature should be measured/shown

### Actual behavior
Tell us what happens instead?
Wrong temp after cold start

### System configuration

Hardware:
NodeMCU+DS18B20 on GPIO2 (D4) , pullup 4k7

Software or git version:
mega-20180104 (binary)

Additional info. Same HW setup performs OK with simple one-wire example arduino sketch (just to verifiy the HW)

ShardanX, 2018-01-13:
Just as a hint for the Devs:
The reading "85° usually says the DS18B20 has a problem with transferring data. 
Maybe the plugin starts too fast reading data after coldboot? For a test give the DS18B20 some "settling time" if possible.

psy0rz, 2018-01-15:
Perhaps this surfaced because we improved startup time, to consume power a long while ago. Before ESPEasy would wait for each device's delaytime before sending data. (after powerup). But since the change, all device will send their data instantly after bootup.

psy0rz, 2018-01-15:
can confim this on the mega branch but not on v2.0, probably because of startup speed difference:

v2.0:
```
5151 : DS : Temperature: 19.06 (28-ee-f2-47-28-16-1-ce)
```

mega:
```
4345 : DS : Temperature: 85.00 (28-ee-f2-47-28-16-1-ce)
```

so we're probably faster than the ds18b20 which is good ;)

TD-er, 2018-01-15:
Even if it doesn't seem to appear to be affected in the v2.0 branch, we should fix it in v2.0
Proper error handling is good to have in any version :)

psy0rz, 2018-01-15:
yeah ill investigate it further and make a proper solutionn

psy0rz, 2018-01-15:
added a delay to setup() just to test, but the problem was still there. there is a difference between p004 in mega and v2.0.

psy0rz, 2018-01-16:
i think the big @BadWolf42 ate it and broke it in https://github.com/letscontrolit/ESPEasy/pull/488 ;)

psy0rz, 2018-01-16:
i discovered another bug: The resolution isnt saved, so after a powercycle the resolution will always be 12bit. Why would we want lower resolution anyway? At the maximum of 12 bit its still only ~700ms, more than fast enough for us. I'm commenting out the resoluation stuff for now.

psy0rz, 2018-01-16:
a simple             Plugin_004_DS_startConvertion(addr); in PLUGIN_INIT doesnt seem to work. (also tried in combination with a readTemp and delays)

time to sleep myself. :)

Oehoeboeroe, 2018-01-16:
I think the scratch pad (0xBE) is read even before the temp conversion (0x44) is requested... Next time the plugin is called the temp of the previous temp conversion is read. 
So in case PLUGIN_READ: first call Plugin_004_DS_startConvertion, then call to Plugin_004_DS_readTemp.
Besides that Plugin_004_DS_startConvertion doesn't take into account any delay (wait for conversion finished... can be up to 750ms).
just my 2 cents...

BadWolf42, 2018-01-16:
Hello @Oehoeboeroe,

My bad... sorry, you are right : Plugin_004_DS_startConvertion should be called before first call to Plugin_004_DS_readTemp. When I rewrote this plugin I forgot to add a PLUGIN_INIT case in Plugin_004.

This should do the trick (in Plugin_004 function, before `case PLUGIN_READ:`) :
```
    case PLUGIN_INIT:
    {
        Plugin_004_DallasPin = Settings.TaskDevicePin1[event->TaskIndex];
        String log = F("INIT : DS ");
        log += Plugin_004_DallasPin;
        addLog(LOG_LEVEL_INFO, log);
        
        if (ExtraTaskSettings.TaskDevicePluginConfigLong[0] != 0){
            uint8_t addr[8];
            // Load ROM address from tasksettings
            LoadTaskSettings(event->TaskIndex);
            for (byte x = 0; x < 8; x++)
                addr[x] = ExtraTaskSettings.TaskDevicePluginConfigLong[x];

            Plugin_004_DS_startConvertion(addr);
            // Call to Plugin_004_DS_readTemp may be < 1s so we add a 800ms delay
            delay(800);
        }
        break;
    }
```

Do you want me to add a pull request to fix ?
Really sorry...

BadWolf42, 2018-01-16:
To details a bit more, the main goal of #488 was to avoid a delay of 750ms for each PLUGIN_READ. This is done by reading the last value then re-starting conversion.

Call to PLUGIN_READ section cannot happen more often than every second, thus the 750ms conversion delay should be ensured. If this is subject to change, a protection must be implemented : test if the last conversion occurred more than 750ms ago then allow read.

TD-er, 2018-01-16:
Another option is to add a part to the more frequently accessed section (N_times_per_second) and determine if it is just a few seconds (or a second) before the plugin will be queried.
The issue a read command and continue working. When the plugin is accessed in its round robin schedule, read the value.
The delay can also be read from within the plugin code.

See for inspiration one of the keypad plugins: https://github.com/letscontrolit/ESPEasy/blob/dfb5b8a4c55033015bb39311ff5b0a8affc813da/src/_P061_KeyPad.ino#L142

psy0rz, 2018-01-16:
That was the first thing i tried but i forgot to get the pin i think. No PR needed since i already started. Thanks!

luciust, 2018-01-16:
Confirmed on mega-20180114-dev (binary release)

Oehoeboeroe, 2018-01-16:
Hi @BadWolf42,
No need to apologize, things happen... and my stuff is not that mission-critical ;-)
Glad I can add a little to the solution.
/BR

commit f45c7650 (2018-01-16): fix #719 (initial ds18b20 value after power on 85 degrees)
closed 2018-01-16 by psy0rz