letscontrolit/ESPEasy #721: Leftover settings when erasing and re-creating a device

s0170071, 2018-01-13:
often, if I play around with the devices, there are some settings leftovers. I think the settings should be erased/initialized  if I define a new device.
e.g. Framed display (oh.. not again !?)
Create it, set it up, play with it. Delete it. 
If I create it again, there are already some entries from the previous one. Shouldn't be because those entries would also be stored in the config file.

Core 2.3.0, branch mega.

TD-er, 2018-01-13:
A bit like this suggestion: https://github.com/letscontrolit/ESPEasy/issues/696#issuecomment-357433348

Great minds think alike :)

psy0rz, 2018-01-15:
its not exactly a duplicate of #696. For some things like controllers this is indeed a "feature": Things like usernames and passwords are not reset for convience. 

However in this case it has to do with LoadCustomTaskSettings(). This is can indeed lead to strange crashes when someone switches between plugins that use loadcustomtasksettings().

We need to check all the plugins that use that and make sure they reset to default when they get (re)created.

psy0rz, 2018-01-23:
this was actually a pretty serious issue, but since not many devices/controller use custom settings, it didnt get triggered that much. it will lead to crashes when switching between 2 devices/controllers that both use custom settings.

commit ed4cdc3d (2018-01-23): clear custom task and controller settings when selecting a different device or protocol. replaces #762 . fixes #721
closed 2018-01-23 by psy0rz