letscontrolit/ESPEasy #724: Output device

Grovkillen, 2018-01-14:
If I use generic devices with PWM I cannot get the current state published. With generic high/low devices I can use switch to display this. 

I'd like a generic output device plugin.

Budman1758, 2018-01-14:
I was thinking more than once it would be nice to have a dummy device that you could store text values in. Would that be something that could be a part of this?

Grovkillen, 2018-01-14:
Sure, but I guess that would take more development since the code doesn't allow anything else than float values for the devices.

But yes, that would be great.

Grovkillen, 2018-03-28:
Maybe do it as a plugin? Let it be used for pwm and servo commands as well?

TD-er, 2018-03-28:
We have a Hardware tab where pull-up status of each pin can be defined.
I guess adding a text field per GPIO to define an alias should be there and not in a plugin.
Only thing is, you really have to make sure to use unique names and not conflict with other keywords.
Or there could be some prefix like `PinPumpRelay` where `Pin` is a fixed prefix.

Grovkillen, 2018-03-28:
Ah, that would make it also fix this request #450

But adding a number would not be good if you want to change a name to another gpio.

Grovkillen, 2018-03-28:
I guess unique naming is already a manual process when it comes to task names and value names... No check there as of today.

TD-er, 2018-03-28:
There is an issue for that to warn users when a name is being used already: #882 (which took me 8 minutes to find)

Grovkillen, 2018-03-28:
So that could also be a part of this then.

closed 2019-10-11 by Grovkillen