letscontrolit/ESPEasy #725: ESP Easy fails to build - fails on IRremoteESP8266

balk77, 2018-01-14:
### Steps to reproduce
How can we trigger this problem?

Fresh install of Atom/Platformio on OS X 10.13.2. 
Download latest pre-release, or any pre release of ESP Easy (source code). I've tried with a number of releases, including a source tree I've built successfully last summer.
Build any target

### Expected behavior
ESP Easy should build nicely

### Actual behavior
Build fails as follows. It seems that the version of _IRremoteESP8266_ shipped with ESP Easy (1.2.0) is outdated and not compatible with the version that Platformio downloads of toolchain-xtensa (~1.40802.0). Source did compile after I've deleted _P016_IR.ino and _P035_IRTX.ino from src.

See also this [topic](http://www.letscontrolit.com/forum/viewtopic.php?f=4&t=3993&sid=330301c6e8b8514e4eb59a5e20296651).

```
lib/IRremoteESP8266/IRMitsubishiAC.cpp:80:44: note:   mismatched types 'std::initializer_list<_Tp>' and 'unsigned int'
temp = max(MITSUBISHI_AC_MIN_TEMP, temp);
^
In file included from /Users/sander/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++
/4.8.2/algorithm:61:0,
from /Users/sander/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/Arduino.h:240,

from lib/IRremoteESP8266/IRremoteInt.h:33,
from lib/IRremoteESP8266/IRremoteESP8266.h:35,
from lib/IRremoteESP8266/IRMitsubishiAC.h:2,
from lib/IRremoteESP8266/IRMitsubishiAC.cpp:12:
/Users/sander/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/4.8.2/bits/stl_algoba
se.h:260:5: note: template<class _Tp, class _Compare> const _Tp& std::max(const _Tp&, const _Tp&, _Com
pare)
max(const _Tp& __a, const _Tp& __b, _Compare __comp)
^
/Users/sander/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/4.8.2/bits/stl_algoba
se.h:260:5: note:   template argument deduction/substitution failed:
lib/IRremoteESP8266/IRMitsubishiAC.cpp:80:44: note:   deduced conflicting types for parameter 'const _

Tp' ('unsigned int' and 'uint8_t {aka unsigned char}')
temp = max(MITSUBISHI_AC_MIN_TEMP, temp);
^
In file included from /Users/sander/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++
/4.8.2/algorithm:61:0,
from /Users/sander/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/Arduino.h:240,
from lib/IRremoteESP8266/IRremoteInt.h:33,
from lib/IRremoteESP8266/IRremoteESP8266.h:35,
from lib/IRremoteESP8266/IRMitsubishiAC.h:2,
from lib/IRremoteESP8266/IRMitsubishiAC.cpp:12:
/Users/sander/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/4.8.2/bits/stl_algoba
se.h:216:5: note: template<class _Tp> const _Tp& std::max(const _Tp&, const _Tp&)
max(const _Tp& __a, const _Tp& __b)
^
/Users/sander/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/4.8.2/bits/stl_algoba
se.h:216:5: note:   template argument deduction/substitution failed:
lib/IRremoteESP8266/IRMitsubishiAC.cpp:80:44: note:   deduced conflicting types for parameter 'const _
Tp' ('unsigned int' and 'uint8_t {aka unsigned char}')
temp = max(MITSUBISHI_AC_MIN_TEMP, temp);
^
lib/IRremoteESP8266/IRMitsubishiAC.cpp:81:44: error: no matching function for call to 'min(unsigned in
t, uint8_t&)'
```

### System configuration
Not relevant

Software or git version:
https://github.com/letscontrolit/ESPEasy/releases/tag/mega-20180114

tarkan123, 2018-01-15:
An issue is already open for this  ( #688 ) 

There is workaround, you can add the following lines to /lib/IRemoteESP8266/   IRremoteESP8266.h

#define max(a,b) ((a)>(b)?(a):(b))
#define min(a,b) ((a)<(b)?(a):(b))

This should allow you to compile without error (confirmed by me), read this on the forum a few days back - and I cannot find the thread so I cannot link to it :(  !!!!!

psy0rz, 2018-01-15:
perhaps we can just update IRremoteESP8266 to the latest version and it might be solved?

anarro, 2018-01-16:
Hi, I have a same error with a library IRremoteESP8266, I try update the library in Platformio Library manager... but nothing..

PR #772 (2018-01-25): [Mega] Allow compile with 2.4.0 core library
Last few days, lots of issues reported by users who installed the 2.4.0 core libraries.
e.g. #769 and #763  
This fixes #725 and thus makes #688 less urgent.
It appeared the default PlatformIO Espressif 8266 1.6.0 libs now include the 2.4.0 core libraries.
Now I have to find a way to get my own environment back to the 2.3.0 core libs and see how I can get PlatformIO Espressif 8266 library back to version 1.5.0

One of the main issues is the change in implementation for min() and max().
closed 2018-01-25 by psy0rz