letscontrolit/ESPEasy #727: Proper user interface to set DST rules

TD-er, 2018-01-14:
As noted in issue #569 and PR #726, the current implementation of the DST rules can only be changed from within the source code and require a recompile.
For most ESPeasy users, the current default DST rules are fine.

A number of ideas to set the DST rules:

- Low level set the values in the web user interface (will take about 12 Bytes of settings)
- Create a list of countries which correspond to the DST + UTC offset (takes lot of code and memory)
- Add a smal(ler) list of used DST rules.

From a user friendly perspective, one would like a selector with all the countries in the world.
But that's just too much for the ESP board.
Setting the values by hand is more work, but with links to Wikipedia it should be doable.

Grovkillen, 2018-01-15:
How many DST schemes are there in the world? I cloud make a wiki page with all countries and a DST scheme code /number next to it. The user enter this number in the GUI. A case switch is used and we're all happy?

Only need a (?) - link in the advanced page then.

TD-er, 2018-01-15:
That would also be an option, to leave it to the documentation page.
First I want #710 to be fixed/investigated, before I will add new data to the settings to be stored.

Judging from the Wikipedia page about this, there are not that many different options and some (like in the UK) switch on the same date, but on different hour of the night.
So changing that would only be needed if timing during the night is essential.

Grovkillen, 2018-01-15:
Great, I think that would be the best option since we want stuff like that funneled to the wiki. We cannot explain all inside the GUI. This being one of those things that wiki is better. :smiley:

papperone, 2018-01-15:
there are countries where DST is not applied... and yes in some others start/end day is different...
maybe we can simplify the code having just 3 or 4 setup (no DST, standard DST, something else(1), something else(2))?

TD-er, 2018-01-15:
In the current implementation I submitted in the PR, there is an option to enable/disable DST and an numeric part to set the UTC offset for the standard time (summer time on northern hemisphere) in minutes.
So it is still possible to allow for 30 and 45 minutes offset some countries have.
To add more options, then the settings have to be extended.

Grovkillen, 2018-02-10:
We can close this now can't we?

closed 2018-02-10 by Grovkillen