letscontrolit/ESPEasy #728: MQTT commands not accepted with later releases

Grovkillen, 2018-01-15:
https://www.letscontrolit.com/forum/viewtopic.php?f=6&t=4009

I just tried it and the log is not reporting any incoming commands using openHAB as MQTT controller.

Grovkillen, 2018-01-15:
OOOOOPS... I forgot that the topic is case sensitive. This is not a bug per say but I did find that the controller settings still used the sysname from when I first created the controller (it published "connection lost" to the old sysname topic)... That must be considered a bug?

TD-er, 2018-01-17:
Hmm, PiDome MQTT will also not work, judging the source code.
I will create a proper function to deal with it, just like Domoticz MQTT plugin does.

TD-er, 2018-01-17:
Second big Hmmm when looking at the code... I guess MQTT controllers can only work when being the first controller.
Maybe that should be split into another issue?

TD-er, 2018-01-17:
See Pull Request #737 
Still have to fix the P037 MQTT Import.
That one allows for multiple MQTT, but with the current fix only the first enabled MQTT plugin will be used.

At least it is better than using settings from the 1st controller, even when that isn't a MQTT controller.

psy0rz, 2018-01-18:
that was a known issue i think: Only the first controller can be MQTT. Its documented on the wiki.

TD-er, 2018-01-18:
And now it has changed into "only one controller can be MQTT, or at least the first enabled MQTT will be used" ;)
Maybe, sometimes we can change it into a full implementation, but I am not sure it will be that useful to have multiple MQTT controllers.
At least it will require a lot more work and I am not really sure we have enough memory to spare.

Grovkillen, 2018-01-18:
The report is now on correct topic: `%sysname%/status`

But I have now realized that the bug also apply to the incoming command on `%sysname%/cmd`

It only respond to the old names topic for incoming command.

Grovkillen, 2018-01-18:
Regarding more MQTT controllers, I would suspect that someone could have the use of two different output formats. JSON (as Domoticz use) or plain text (as openHAB use).

But for now I really like that I'm able to have MQTT on other controller number than 1 since many people will have 1 occupied with some (perhaps HTTP) controller before wanting to test MQTT.

Grovkillen, 2018-01-18:
I will :+1:

closed 2018-01-18 by Grovkillen

Grovkillen, 2018-01-18:
Just a quick question, did you change so we are now allowed to have MQTT on controller 1..3?

I added openHAB MQTT on controller 1, removed it, tried to add it to controller 2, it was greyed out.
I added openHAB MQTT on controller 1, replaced it with Domoticz HTTP (disabled), tried to add it to controller 2, it was greyed out.

TD-er, 2018-01-18:
Hmm, grey-out is something done in the webinterface. Haven't looked into that.
In the code it now either uses the set controller in the plugin, or when no plugin event is available, it searches for the first enabled MQTT controller when dealing with MQTT things.
It sounds like there was already some protection built in to disallow it.

Grovkillen, 2018-01-18:
Should I open an issue for that :)