letscontrolit/ESPEasy #730: Store free text to be requested using URL

Grovkillen, 2018-01-16:
Currently I use rule page 4 to store free text information regarding my units. I just add // and comment a bunch of stuff.

Ideally I would have a page kinda like the JSON page where I could store my information as JSON and then I could only fetch all these JSON strings and parse them centrally. A nice feature to add to that would be if the units settings would be part of that JSON code as well.

URL could be: /jsondump or maybe /jsonhook or something else awesome! :)

TD-er, 2018-01-16:
Features to add along with this:

- explicitly select to export passwords (not paste passwords by accident on a forum)
- Import json settings
- partly export settings
- partly import settings
- backport the json-export to older version of ESPeasy to ease upgrading
- store json on the spiffs as a backup/undo (multiple versions?)
- Allow undo of settings per plugin/controller/rule
- Allow clear settings per plugin/controller/rule

Grovkillen, 2018-01-17:
Possibly also add the bin file name if OTA flashed.

TD-er, 2018-01-18:
I've looked into how the settings are being stored and I think it should be fairly easy to generate Json from the settings to allow to export/import the settings and perhaps also export the old versions.

Then we could add some Python code to convert settings if they change later on.
So maybe change the topic, or should I create a new one to address the JSON import/export?
This also allows other issues which need to extend some settings which is quite impossible in the current setup.

Grovkillen, 2018-01-19:
Wow, that is so awesome! We should make a new issue for the export /import

Grovkillen, 2018-10-23:
#572 is closely related and can be closed.