letscontrolit/ESPEasy #732: P029 MQTT helper works when disabled

wdonker, 2018-01-17:
**NOTE: This is not a support forum! For questions and support go here: https://www.letscontrolit.com/forum/viewforum.php?f=1**

### Steps to reproduce
Add a device Domoticz MQTT Helper (P029) and uncheck 'enabled'.

Does the problem presist after powering off and on? (just resetting isnt enough sometimes)
Yes

### Expected behavior
Device should not switch it's GPIO since it is not enabled.

### Actual behavior
Device does switch it's GPIO.

### System configuration

Hardware:

ESP8266 NodeMCU

Software or git version:
v2.0-20180112

wdonker, 2018-01-17:
See: [http://www.letscontrolit.com/forum/viewtopic.php?f=6&t=4016&p=22726#p22726](http://www.letscontrolit.com/forum/viewtopic.php?f=6&t=4016&p=22726#p22726)

TD-er, 2018-01-18:
With Pull Request #737 (for issue #728 ) I also fixed some silly things with the MQTT controllers.
Could you please check if this issue is also closed by this fix?

wdonker, 2018-01-18:
Sorry, it did not.
I can not compile ESPEasy myself yet but I think this bug is in  _C002.ino (a check on TaskDeviceEnabled must be added near line 86).

TD-er, 2018-01-18:
OK, I will have a look at it, this evening.
https://github.com/letscontrolit/ESPEasy/blob/5821377b6c6e90154ac6c42a86862460da141740/src/_C002.ino#L86

TD-er, 2018-01-18:
Let's hope @psy0rz will merge my pull req. 
Please test when a new build is made (with this PR in it)

psy0rz, 2018-01-21:
fixed by @TD-er

closed 2018-01-21 by psy0rz

TD-er, 2018-01-22:
Do you have an example of log output you had with older versions?

wdonker, 2018-01-22:
Hi  Gijs,

The line I mentioned in the issue is missing now.

Grtz, Wim