letscontrolit/ESPEasy #733: ESPEasy MQTT import negative values

Securond, 2018-01-17:
Did not get to take negative values from the topic. The result is always 0.00.
(firmware ESPEasy_mega-20180117.bin)

TD-er, 2018-01-17:
Can you please elaborate on what is actually the issue here?
Also examples and steps to reproduce + what is expected would be great.

Securond, 2018-01-17:
I need to output negative temperature values ​​from the broker MQTT to the LCD display screen. positive values ​​are output without problems. negative values ​​are not output and always show 0.0 Here there are screenshots - http://forum.amperka.ru/threads/espeasy.14196/

Grovkillen, 2018-01-17:
I can confirm this.

Grovkillen, 2018-01-17:
Negative float values are not accepted. Value turn to zero.

TD-er, 2018-01-17:
OK then we will look into it.

TD-er, 2018-01-17:
Ah that looks like an easy fix.
I am working on it :)

uzi18, 2018-01-17:
Someone wrote last time issue post about wrong mqtt values, same problem?

Securond, 2018-01-17:
Thank you “TD-er”.  We are waiting for the next firmware.

TD-er, 2018-01-17:
@psy0rz now has to merge it and then around 5 am tomorrow the new build will be ready.

N.B. just did a compile test, not tested with a MQTT topic.
So please test it :)

psy0rz, 2018-01-17:
let us know if its fixed. next build will be avaiable in a few ours.

Grovkillen, 2018-01-18:
It's working as intended. Thanks @TD-er !! Must be a record time. :smile: 

#735

closed 2018-01-18 by Grovkillen

Securond, 2018-01-18:
Thank you, negative values are now displayed without problems. But!!! Stopped working, time display on LCD screen %-(( :
%systime% - works
%syshour% - does not work
%sysmin% - does not work
%syssec% - does not work
%sysday% - does not work
%sysmonth% - does not work
%sysyear% - does not work
%sysyears% - does not work
%ip% - works
%uptime%  - works

Securond, 2018-01-18:
https://www.letscontrolit.com/wiki/index.php/ESPEasy_System_Variables

Grovkillen, 2018-01-18:
#738 is not confirmed by me. I have tested these variables internally on latest mega release with no problem what so ever. The only thing I haven't been able to test is on a actual LCD but that seems like a far fetched issue (but you never know).