letscontrolit/ESPEasy #734: ESP Mega - timerSet & TaskValueSet not work with"7-segment display"

newRainbow, 2018-01-17:
Hello!
There is a problem - When the "7-segment display" plugin is aktivated,
the timerSet & TaskValueSet commands stop working.
These commands are simply ignored.

Example:

on System#Boot do
TaskValueSet,5,1,10
timerSet,1,3
endon
on Rules#Timer=1 do
Gpio,2,1
timerSet,2,3
endon
on Rules#Timer=2 do
Gpio,2,0
timerSet,1,3
endon

WeMos d1 mini 
Build | 20100 - Mega (core 2_3_0)
GIT version | mega-20180117

TD-er, 2018-01-17:
The related topic on the forum: https://www.letscontrolit.com/forum/viewtopic.php?p=22740#p22740

newRainbow, 2018-01-18:
If plugin  "7-segment display" is disabled, the example works. And if you activate - then no.
It is not necessary to connect the display, just plug in the plugin.

Grovkillen, 2018-03-16:
Still an issue?

closed 2018-03-16 by Grovkillen