letscontrolit/ESPEasy #738: %sysday% %sysmonth% ......... - not working

Securond, 2018-01-18:
Stopped working, time display on LCD screen %-(( :
%systime% - works
%syshour% - does not work
%sysmin% - does not work
%syssec% - does not work
%sysday% - does not work
%sysmonth% - does not work
%sysyear% - does not work
%sysyears% - does not work
%ip% - works
%uptime% - works

https://www.letscontrolit.com/wiki/index.php/ESPEasy_System_Variables

Grovkillen, 2018-01-18:
I cannot confirm this. Internally these variables are working just fine with latest mega (20180118). But I have not tested with LCD. Could you please test a bit more and also on another unit?

Securond, 2018-01-18:
@Grovkillen I'm sorry, I am a fool. Downloaded the wrong firmware.

Grovkillen, 2018-01-18:
I happens to all of us from time to time :roll_eyes: :+1:

closed 2018-01-18 by Grovkillen

Securond, 2018-01-18:
@Grovkillen Thanks again for Your speed work !!!

Grovkillen, 2018-01-18:
Don't mention it, I only do testing and wiki. @TD-er did one heck of a sprint with your bug :)

Securond, 2018-01-18:
I give thanks to @TD-er from Russian users )))

psy0rz, 2018-01-19:
so cool that ESPEasy is also used in russia! :)