letscontrolit/ESPEasy #739: If changed sysname the unit will only listen to cmd on "old sysname topic"/cmd

Grovkillen, 2018-01-18:
The report is now on correct topic: %sysname%/status

But I have now realized that the bug also apply to the incoming command on %sysname%/cmd

It only respond to the old names topic for incoming command.

Related to #728

TD-er, 2018-01-29:
Can you please explain what is needed here?
Should we force a MQTT client disconnect/reconnect/subscribe when the system name has changed?

Grovkillen, 2018-01-29:
Yes thats exactly my wish :+1:

Grovkillen, 2018-01-29:
And a thought I had last night... we will default give the unit the sysname "ESP_Easy", will this mess up the DNS? I mean, sometimes old IP numbers are cached locally to that host name, leading to strange behavior if the sysname is used on a new unit.

Not related to this one but just a thought :)

psy0rz, 2018-02-01:
@Grovkillen @TD-er  probably fixed it in that PR. reopen it if its not working like expected :)

closed 2018-02-01 by psy0rz

Grovkillen, 2018-02-02:
Worked like a charm!! :)