letscontrolit/ESPEasy #745: Deleting Syslog Server IP address result in a IP number starting with "0"

Grovkillen, 2018-01-19:
If I enter a IP number for my syslog server, save and then remove the address (deleting it) and then save again the IP is not removed but changed to a IP number starting with 1st octet "0".

I.E:
192.168.0.170 --> 0.168.0.170

In order to fully delete the IP I need to enter the IP number 0.0.0.0

TD-er, 2018-01-19:
Sounds like something that could occur on more places.

psy0rz, 2018-01-19:
yeah i also seen it with a controller yesterday i think. i assumed i made a mistake since it was 4:30, but its probably an underlying bug.

also ipadresses in the logs are shown as numbers. (probably need to call .toStr() or something on an IPAdress class)

Budman1758, 2018-01-19:
I can confirm this.  Just happened to me.

psy0rz, 2018-01-21:
fixed

closed 2018-01-21 by psy0rz