letscontrolit/ESPEasy #746: I ask You for some help. Is it possible, to add new SSD1306 addresses 0x78 (0x7A) ? I have many OLED displays with this address, and there is many such displays on Aliexpress

FreddyPL, 2018-01-19:
**NOTE: This is not a support forum! For questions and support go here: https://www.letscontrolit.com/forum/viewforum.php?f=1**

### Steps to reproduce
How can we trigger this problem?

Does the problem presist after powering off and on? (just resetting isnt enough sometimes)

### Expected behavior
Tell us what should happen?

### Actual behavior
Tell us what happens instead?

### System configuration

Hardware:

Software or git version:

TD-er, 2018-01-19:
Link to forum topic: https://www.letscontrolit.com/forum/viewtopic.php?p=22817#p22817

psy0rz, 2018-01-24:
yeah this is what surprised me as well: although it says 0x78 on the PCB, the 0x3c address is actually working. see this comment: https://www.letscontrolit.com/forum/viewtopic.php?p=22817#p22818

closed 2018-01-24 by psy0rz