letscontrolit/ESPEasy #748: PUYA flash chip does not have SPIFFS available

ondabeach, 2018-01-20:
It appears that the PUYA flash chip is being adopted by a growing number of manufacturers of the esp8266 modules. My most recent purchases of esp-01 modules from 3 separate suppliers have all had the PUYA chip on-board (for a total of 95 modules). The issue is that Arduino IDE x.xx sketches using any 8266 core can not currently write to SPIFFS on modules with this chip.

**NOTE** that all other functions are fine with this chip, as far as I have tested so far it is only SPIFFS that is a affected. Also worth noting is that the IDE's "Sketch Data Upload" function can successfully write to SPIFFS including file contents and sketches can then read the file contents, sketches just can't write to SPIFFS. Furthermore you can create a file in so far as that DIR can see the file and list the file name, but the file will have no contents.

I'd like to know how many others have bought esp modules that have the PUYA branded chips on board as I feel that this could soon be a very widespread problem.

**Please don't post 'things to try for a fix' here.** I'd rather only hear from those that actually have modules with the PUYA flash chip so I can get an idea of how widely they are being adopted.

Moderators, please don't close this issue without giving others with this chip time to respond. Rather, point them here. The other threads relating to this have become bloated with suggestions for fixes etc., so I think a thread that is only focused on those with the PUYA chip is the best way to gather info for the key guys to solve this.

psy0rz, 2018-01-20:
this is duplicate of #650 i think

ondabeach, 2018-01-20:
@psy0rz there are at least 3 current threads that are related to this issue but they are all getting bloated by suggestions from people that don't even have a PUYA chip to test with. I know they, and you are trying to help but...

**PLEASE, only people with PUYA chipped modules, or the 8266 'gurus' that are actively working on the problem to comment here**

psy0rz, 2018-01-20:
ok thanks i'll leave it here for now util we have a fix :)

TD-er, 2018-01-20:
So just to make it clear, it is just for a rough impact analysis?
I do have some ideas on what might be causing this, but I do not have any of these chips yet to test with.

ondabeach, 2018-01-20:
No theories please! IF YOU DON'T HAVE A PUYA CHIPPED MODULE THEN PLEASE DON'T  POST HERE!

psy0rz, 2018-01-20:
@ondabeach td-er is one of the main developers of ESPEasy at this moment. ;)

ondabeach, 2018-01-20:
Without a module to test with there is no point in commenting here. There are already 3 threads full of theories and ideas of 'things to try'. I was trying to avoid making this number 4. I don't know how I could have made that any more clear.

riker65, 2018-01-21:
Hi
have the same problem - esp easy not starting on puya.esp-01s
It started once r147, could assign a SSID, but did not restart after that.
did a new flash but now espeasy  is not restarting any more.
can still flash normal sketch without spiffs usage.

[image]

ondabeach, 2018-01-21:
@riker65 that's definitely the problem chip you have there. I see that horrid PUYA symbol in my sleep ;)

I'm having to replace the flash chips so I can make my products and keep my business going until @igrr or someone can find a solution. I'm using the Adesto AT25SF041-SSHD-B which is only a 512Kbyte chip but is all my sketch requires. I get them from RS Components but other places like mouser.com have them. I assume that the 1Mbyte and bigger chips from the same family would also support SPIFFS too.

Hopefully it won't be too long before one of the guns around here get their hands on a PUYA chipped module so they can actually fix the problem rather than take stabs in the dark. The replacement flash chips are only costing me 40cents delivered but it's a pain in the butt having to replace them.

riker65, 2018-01-21:
Hi 
just check the other I am having. Also Puya, but there at least r147 is running.
Strange
[image]

ondabeach, 2018-01-21:
The issue with the PUYA chip is that you can't write to SPIFFS with a sketch. What is your sketch storing in SPIFFS?

riker65, 2018-01-21:
I do place some website xml on spiffs
have to change my design so.

ondabeach, 2018-01-21:
Ok, keep an out here for a fix.

psy0rz, 2018-01-28:
i received my ESP-01's, bought them at the same ebay address that someone mentioned. I'll test and try to debug now.

TD-er, 2018-01-28:
I also received one by mail, sent by a user. (Tim Karnhof)
Hopefully we can test with the SPIFFS_NO_BLIND_WRITES flag mentioned here: https://github.com/esp8266/Arduino/issues/4061#issuecomment-359654138

ondabeach, 2018-01-28:
@TD-er that's great. I'll keep my fingers crossed that it works. I need to replace another batch of the PUYA chips on my modules so I can keep making my products. If it does work I'd love a link where I can download the tweaked files to try. Are you tweaking 2.3 core files? 2.4 core seems to have too many bugs still to be used in a commercial product.

psy0rz, 2018-01-30:
Did someone confirm that the ESP-01S without PUYA chip does operate correctly?

I tried with SPIFFS_NO_BLIND_WRITES, but no difference here. (see https://github.com/esp8266/Arduino/issues/4061#issuecomment-361428719)

ondabeach, 2018-01-30:
@psy0rz it's looking pretty scary at this stage. No-one in the brains trust seems to be making any headway with this :(

csipeter, 2018-02-03:
Hello!
These boards have P25Q80H soldered on it. Did someone read that chip's datasheet? I downloaded it from Puya Semiconductor's site. So I share it here. Hope this helps.
[PUYA-P25Q80H.pdf](https://github.com/letscontrolit/ESPEasy/files/1691463/PUYA-P25Q80H.pdf)

ondabeach, 2018-02-03:
Hi @csipeter , the brains trust do have the datasheet. I downloaded it before their website went off the air a few weeks ago. Good to see it is back up again, we were worried they had gone out of business.

Unfortunately the datasheet hasn't helped them solve the problem however :(

ondabeach, 2018-02-03:
@igrr, @TD-er the Puya website is back on line :) Maybe you can ask them for help.

http://www.puyasemi.com/English/index.php

elucas2uy, 2018-02-07:
Is this related too?..  but with other chip than the PUYA one?

https://www.letscontrolit.com/forum/viewtopic.php?f=5&t=4803

ondabeach, 2018-02-08:
Possibly. @TD-er  is the one to ask, he is one of the owners of ESPEasy I believe

TD-er, 2018-02-08:
@ondabeach
I am not an owner, just an enthusiastic developer ;)

ondabeach, 2018-02-08:
@TD-er, but apparently not enthusiastic enough to answer @elucas2uy's enquiry?

s0170071, 2018-02-08:
@ondabeach, rest assured, everybody here wants to help. We all do it just for the fun of it and the occasional thumbs up. And so does @TD-er.  Judging from his acitivities around here he looks pretty enthusiastic to me. Maybe that's why you believe that he owns this repo ;-)

If you feel like you have been ignored, just ask again for an answer. Politely if you like.  It was probably not intentional.
Challenging someones committment, however,  only works for little children but annoys grown ups. A lot.

So let's keep it professional, shall we ?

ondabeach, 2018-02-08:
s0170071, please don't comment after only reading one or two entries in a thread. If you had absorbed what you'd read instead of 'jumping to the rescue' of someone you see as a peer, you'd know that I said I believed him to be a co-owner of ESPEasy... not this repo. I was actually told by one of your more senior peers that TD was the co-owner of ESPEasy.

Similarly, if TD had bothered to read back as far as two posts before getting his back up, he would have seen the question that was asked. He's had his nose out of joint since I poked a little fun at him for suggesting that the PUYA flash chip problem might be due to the fact that the esp-01 module doesn't have an RF shield. And no, I'm not kidding, he really did.

I have asked, politely, repeatedly, on numerous threads, of numerous people, including contributors of God like stature, but all to no avail. And I have literally been ignored, and no, not since, but long before I started getting pissed at the lack of action on what is a serious problem to anyone who wants to use SPIFFS on an esp-01 module.

Yes, I understand fully the volunteer basis of this repo, but to simply walk away and bury your head in something else without any kind of feedback is anything but professional. If they don't know how to fix it or even what the problem is, just say so. And before you bother replying, spend a bit of time researching this issue, and by a little time I don't mean ten minutes.

I think it might actually be less painful to replace the flash chip on every module I use, rather than to keep getting my hopes up of anyone around here focusing on the issue long enough to resolve it.

Maybe I'll update when core 6.0 is released and see if any headway has been made.

Grovkillen, 2018-02-08:
No need to close this issue, we have some units now where we try to get the SPIFFS reading/writing working... but PUYA need some other approach than what is currently used in the ESPlibrary. Thanks for your feedback. :+1:

ondabeach, 2018-02-08:
Thanks Grovkillen, a little feedback goes a long way. I've emailed PUYA and suggested it's in their best interest to work with the github 8266 repos to resolve this. I've also let guys on the 8266/Arduino repo know that Puya Semi's website is back up so they can contact them. I have no idea if they've done anything or if they've even read my posts. It's a bit like throwing stones down a bottomless well at times.

Grovkillen, 2018-02-08:
Yes, we need to add as much pressure as possible to PUYA. Still not a word from them, I guess we're not in the big leagues...

s0170071, 2018-02-08:
LOL. I just went to the puya site. They call their SPI flash "indutrial". 
http://www.puyasemi.com/English/pro.php?SortID=2&ID=30

csipeter, 2018-02-08:
Hehehe. In du TRIAL.

beicnet, 2018-02-08:
...like they PUYA chip! :rofl:

ondabeach, 2018-02-09:
@Grovkillen, exactly, we're tiny little fish that don't rate. I'm not expecting to hear back from them either.

Like it or not, the whole world of Arduino and even the world of IOT at large don't mean a damn thing to a chip manufacturer. They sell their products by the thousands, tens and hundreds of thousands to companies that manufacture commercial devices such as set top boxes, smart fridges, microwave ovens and thousands of other types of gadgets and products.

There are probably two or more middlemen between PUYA and the little companies that make our modules like the generic esp-01.

Quite possibly companies like Adafruit buy directly from chip suppliers, but either way, the volumes are insignificant compared to the big fish like Sony, Philips, LG and so on.

In short, they're not really going to care about us. We need to face that and look for a solution ourselves. Bitching about it won't get us anywhere.

The suppliers that sell the generic esp-01's on eBay or AliExpress are going to buy from the cheapest provider regardless, and at the moment it looks like that means a module with a PUYA chip on board. The fact that the modules work except for SPIFFS writing means they can get away with selling them because for most sketches they work.

Most people will just throw a $3 module in the bin rather than stuff around sending it back. If 1 in 20 people pursue a refund, the suppliers are happy to live with that. Don't forget that a lot of ebay suppliers of these things are just individual dudes working out of their mum's garage.

So, the options are either fork out for an Adafruit, Saprkfun or whatever module where you'll get something that you know will work, or buy a cheapie and be prepared to spend another 40 cents and replace the flash chip.

It sucks balls, but hey, that's life.

ondabeach, 2018-02-09:
@Grovkillen I have great news guys.
@igrr has fixed the SPIFFS write problem with the PUYA flash chip.
See #4061

Grovkillen, 2018-02-09:
Awesome news!!

ondabeach, 2018-02-09:
Yes indeed. I guess this can be closed now and referred over to https://github.com/esp8266/Arduino/issues/4061#

I'll leave that up to you @Grovkillen, I'm hitting the hay.

Grovkillen, 2018-02-09:
I close and we'll wait for the merge.

closed 2018-02-09 by Grovkillen

adapso, 2019-02-18:
I no longer know what to do, I have already followed all the steps described in the comments and I can not make it work. I have a 20 ESP-01 with PUYA chip.

TD-er, 2019-02-18:
@adapso 
If you're building using PlatformIO, then please have a look at the platformio.ini we use here at ESPeasy.
I patched the sources for core 2.3.0, 2.4.0, 2.4.1 and 2.4.2 (the patch is now included in core 2.5.0)
So for platformio, you can use the platform defines we use:
https://github.com/letscontrolit/ESPEasy/blob/a8bfa71bd5a6ddd6710fffb5445dfd1cab1e3b94/platformio.ini#L93-L119

For Arduino IDE, the Puya patch should be included in the core 2.5.0 and you must enable this patch by setting the define `PUYA_SUPPORT=1` at compile time. (probably adding something like `-DPUYA_SUPPORT=1` or some menu option)

adapso, 2019-02-19:
> @adapso
> Se você está construindo usando o PlatformIO, então dê uma olhada no platformio.ini que usamos aqui no ESPeasy. 
> Eu patchei os fontes para o núcleo 2.3.0, 2.4.0, 2.4.1 e 2.4.2 (o patch agora está incluído no núcleo 2.5.0) 
> Então, para o platformio, você pode usar a plataforma define que usamos:
> [ESPEasy / platformio.ini](https://github.com/letscontrolit/ESPEasy/blob/a8bfa71bd5a6ddd6710fffb5445dfd1cab1e3b94/platformio.ini#L93-L119)
> 
> Linhas 93 a 119 em [a8bfa71](/letscontrolit/ESPEasy/commit/a8bfa71bd5a6ddd6710fffb5445dfd1cab1e3b94)
> 
>  [core_2_3_0] 
>  plataforma                   = https://github.com/TD-er/platform-espressif8266.git # patch / v1.5.0_Puya 
>  build_flags                = -D BUILD_GIT = ' "$ {sysenv.TRAVIS_TAG}" ' 
>                              -DNDEBUG 
>                              -DVTABLES_IN_FLASH 
>                              -fno-exceptions 
>                              - DWEBSERVER_RULES_DEBUG = 0
>                              - DPUYA_SUPPORT = 1
>                              -DARDUINO_ESP8266_RELEASE_2_3_0 
>   
>  [core_2_4_0] 
>  plataforma                   = https://github.com/TD-er/platform-espressif8266.git # patch / v1.6.0_Puya 
>  build_flags                = $ {esp82xx_defaults.build_flags} -DARDUINO_ESP8266_RELEASE_2_4_0
>   
>  [core_2_4_1] 
>  plataforma                   = https://github.com/TD-er/platform-espressif8266.git # patch / v1.7.3_Puya 
>  build_flags                = $ {esp82xx_defaults.build_flags} -DARDUINO_ESP8266_RELEASE_2_4_1
>   
>  [core_2_4_2] 
>  plataforma                   = https://github.com/TD-er/platform-espressif8266.git # patch / v1.8.0_Puya 
>  build_flags                = $ {esp82xx_defaults.build_flags}
>   
>  [core_2_5_0] 
>  ; plataforma = platform-espressif8266@1.9.0 
>  ; plataforma = https://github.com/TD-er/platform-espressif8266.git#feature/2.5.0-beta1 
>  plataforma                   = https://github.com/platformio/platform-espressif8266.git # feature / stage 
>  build_flags                = $ {esp82xx_2_5_0.build_flags}
> Para o Arduino IDE, o patch Puya deve ser incluído no núcleo 2.5.0 e você deve habilitar este patch definindo a definição `PUYA_SUPPORT=1`em tempo de compilação. (provavelmente adicionando algo como `-DPUYA_SUPPORT=1`ou alguma opção de menu)

I am using Arduino IDE, I upgraded the core to 2.5.0 and it still does not work.

TD-er, 2019-02-19:
Have you set the define to enable `PUYA_SUPPORT` ?
And where did you enable it?
You must make sure to do a clean build, since it must be triggered to re-build the core libraries.

21h, 2019-03-10:
2.5.0
`#define PUYA_SUPPORT 1 `
no effect

TD-er, 2019-03-10:
Where do you define that?
You have to make sure it is already defined before the core libraries are being built, so that's why I set it as defines for the build command itself. It is not set as a define in my own source code.

21h, 2019-03-11:
I defined it in head of my program. 

In core sources .h file I see define it with it not defined before with value 0. 

In .c file I see checking only existance of that define (remember, it was defined in .h file if not defined before).

Its very strange. Looks like PUYA_SUPPORT not programmed correctly in latest 2.5.0.

TD-er, 2019-03-11:
No, that's not where you need to put it, you need to give it as an argument to the compiler.
So in PlatformIO it is `-DPUYA_SUPPORT=1` in the list of build flags.
I don't know what you need for the Arduino IDE.

You must also make sure to make a full clean build then.

PUYA support is working fine, but you have to enable the flag in the correct place or else it will not be active when the core libraries are built.

adapso, 2019-03-11:
I already gave up, I could not make it work, my project is stopped because of this PUYA. I try to make new purchases on Ebay and Aliexpress and just send me this damn PUYA

TD-er, 2019-03-11:
Are you using PlatformIO or Arduino IDE?

uzi18, 2019-03-11:
I don't understand why it is not default

adapso, 2019-03-12:
> Are you using PlatformIO or Arduino IDE?

Arduino IDE

TD-er, 2019-03-12:
> I don't understand why it is not default

That's what the guys from the esp8266/Arduino repo requested.

In the Arduino IDE, you can setup some stuff like board settings and such. Isn't there now an option to enable the PUYA patch? They discussed about adding that to the options.
Or maybe some build flags you can add somewhere?

Looks like you may have to look for a line like this:
`compiler.c.flags=` in `platform.txt` and then add the define `-DPUYA_SUPPORT=1` to that line (not sure if it will be in multiple places)

adapso, 2019-03-21:
> > @adapso 
> > Se você está construindo usando o PlatformIO, então dê uma olhada no platformio.ini que não use no ESPeasy. 
> > Você pode usar as 
> > seguintes fontes para obter o núcleo 2.3.0, 2.4.0, 2.4.1 e 2.4.2 (o patch agora está incluído no núcleo 2.5.0) Então, para o platformio, você pode usar uma plataforma que define o que é: 
> > [ESPEasy / platformio.ini](https://github.com/letscontrolit/ESPEasy/blob/a8bfa71bd5a6ddd6710fffb5445dfd1cab1e3b94/platformio.ini#L93-L119)
> > Linhas 93 a 119 em [a8bfa71](/letscontrolit/ESPEasy/commit/a8bfa71bd5a6ddd6710fffb5445dfd1cab1e3b94)
> > [core_2_3_0] 
> > plataforma = https://github.com/TD-er/platform-espressif8266.git # patch / v1.5.0_Puya 
> > build_flags = -D BUILD_GIT = '"$ {sysenv.TRAVIS_TAG}"' 
> > -DNDEBUG 
> > -DVTABLES_IN_FLASH 
> > - exceções-fno 
> > - DWEBSERVER_RULES_DEBUG = 0 
> > - DPUYA_SUPPORT = 1 
> > -DARDUINO_ESP8266_RELEASE_2_3_0
> > [core_2_4_0] 
> > plataforma = https://github.com/TD-er/platform-espressif8266.git # patch / v1.6.0_Puya 
> > build_flags = $ {esp82xx_defaults.build_flags} -DARDUINO_ESP8266_RELEASE_2_4_0
> > [core_2_4_1] 
> > plataforma = https://github.com/TD-er/platform-espressif8266.git # patch / v1.7.3_Puya 
> > build_flags = $ {esp82xx_defaults.build_flags} -DARDUINO_ESP8266_RELEASE_2_4_1
> > [core_2_4_2] 
> > plataforma = https://github.com/TD-er/platform-espressif8266.git # patch / v1.8.0_Puya 
> > build_flags = $ {esp82xx_defaults.build_flags}
> > [core_2_5_0] 
> > ; plataforma = platform-espressif8266@1.9.0 
> > ; plataforma = https://github.com/TD-er/platform-espressif8266.git#feature/2.5.0-beta1 
> > plataforma = https://github.com/platformio/platform-espressif8266.git # recurso / estágio 
> > build_flags = $ {esp82xx_2_5_0.build_flags} 
> > Para o Arduino IDE, o patch deve ser incluído no núcleo 2.5.0 e você deve habilitar este patch definindo a definição `PUYA_SUPPORT=1`em tempo de compilação. (algo adicionando algo como `-DPUYA_SUPPORT=1`alguma opção de menu)
> 
> Estou usando o Arduino IDE, atualizei o núcleo para 2.5.0 e ainda não funciona.

> @adapso
> Se você está construindo usando o PlatformIO, então dê uma olhada no platformio.ini que usamos aqui no ESPeasy.
> Eu patchei os fontes para o núcleo 2.3.0, 2.4.0, 2.4.1 e 2.4.2 (o patch agora [...]

I installed PlatformIO and now, what I do not understand where to edit DPUYA_SUPPORT = 1

TD-er, 2019-03-21:
I don't thing `plataforma` is a valid entry in the Platformio.ini configuration.
Also the url's do not have spaces.

Just have a look at the config we're using. First the sections describing the core environment and some default flags describing to set the PUYA flag:

https://github.com/letscontrolit/ESPEasy/blob/147da97f24f28e1776c562b4eac5f081cc690388/platformio.ini#L72-L122

You can see the sections labelled with something like `[core_stage]` and `[esp82xx_2_5_0]`
As you can see, the `core_stage` section does refer to `esp82xx_2_5_0` in the `build_flags` parameter.

Now the same for the next important part of this platformio.ini file:

https://github.com/letscontrolit/ESPEasy/blob/147da97f24f28e1776c562b4eac5f081cc690388/platformio.ini#L154-L180

The most important sections here are `common` , `regular_platform` and `beta_platform`
These are used in all other env configurations (the ones listed in PlatformIO to build)

Let's just focus on the `normal` builds we make.

https://github.com/letscontrolit/ESPEasy/blob/147da97f24f28e1776c562b4eac5f081cc690388/platformio.ini#L184-L202

I think you can see the pattern already.
They are just referring to other sections to make sure we do not have a lot of copy/paste configurations in the file and it makes it also easier to compare different builds.

Since you are trying to build for a PUYA node, I guess you have an ESP8266 + 1M flash, so just take that config:

https://github.com/letscontrolit/ESPEasy/blob/147da97f24f28e1776c562b4eac5f081cc690388/platformio.ini#L252-L258

And now we define the build for a node with "normal" build and 1M flash:

https://github.com/letscontrolit/ESPEasy/blob/147da97f24f28e1776c562b4eac5f081cc690388/platformio.ini#L362-L378

As you can see these are only references to sections we made before.

But the most important part you need to know is that it all comes down to adding the PUYA flag to the `build_flags`.
So that list of flags will have a lot of parameters and one of them is the `PUYA_SUPPORT=1`
Flags set in that list must have `-D` in front of them, so either `-D PUYA_SUPPORT=1` or `-DPUYA_SUPPORT=1`.
So the `build_flags` line will be something like this:
```
build_flags               = ....lots_of_other_flags...  -DPUYA_SUPPORT=1
```

I hope it is now clear how this works.

adapso, 2019-03-21:
> Eu não coisa `plataforma`é uma entrada válida na configuração Platformio.ini. 
> Além disso, o URL não possui espaços.
> 
> Basta dar uma olhada na configuração que estamos usando. Primeiro, as seções que descrevem o ambiente principal e alguns sinalizadores padrão descrevendo para definir o sinalizador PUYA:
> 
> [ESPEasy / platformio.ini](https://github.com/letscontrolit/ESPEasy/blob/147da97f24f28e1776c562b4eac5f081cc690388/platformio.ini#L72-L122)
> 
> Linhas 72 a 122 em [147da97](/letscontrolit/ESPEasy/commit/147da97f24f28e1776c562b4eac5f081cc690388)
> 
>  [esp82xx_defaults] 
>  build_flags                = -D BUILD_GIT = ' "$ {sysenv.TRAVIS_TAG}" ' 
>                              -D NDEBUG 
>                              -lstdc ++ -lsupc ++ 
>                              -mtarget-align 
>                              -DPIO_FRAMEWORK_ARDUINO_LWIP2_LOW_MEMORY 
>                              - DWEBSERVER_RULES_DEBUG = 0
>                              -DVTABLES_IN_FLASH 
>                              - DPUYA_SUPPORT = 1
>   
>  [esp82xx_2_5_0] 
>  build_flags                = -D BUILD_GIT = ' "$ {sysenv.TRAVIS_TAG}" ' 
>                              -DNDEBUG 
>                              -DVTABLES_IN_FLASH 
>                              -fno-exceptions 
>                              -DPIO_FRAMEWORK_ARDUINO_LWIP2_LOW_MEMORY_LOW_FLASH 
>                              - DWEBSERVER_RULES_DEBUG = 0
>                              - DPUYA_SUPPORT = 1
>                              -DCORE_POST_2_5_0 
>   
>   
>  [core_2_3_0] 
>  plataforma                   = https://github.com/TD-er/platform-espressif8266.git # patch / v1.5.0_Puya 
>  build_flags                = -D BUILD_GIT = ' "$ {sysenv.TRAVIS_TAG}" ' 
>                              -DNDEBUG 
>                              -DVTABLES_IN_FLASH 
>                              -fno-exceptions 
>                              - DWEBSERVER_RULES_DEBUG = 0
>                              - DPUYA_SUPPORT = 1
>                              -DARDUINO_ESP8266_RELEASE_2_3_0 
>   
>  [core_2_4_0] 
>  platform                  = https://github.com/TD-er/platform-espressif8266.git#patch/v1.6.0_Puya 
>  build_flags               = ${esp82xx_defaults.build_flags} -DARDUINO_ESP8266_RELEASE_2_4_0 
>   
>  [core_2_4_1] 
>  platform                  = https://github.com/TD-er/platform-espressif8266.git#patch/v1.7.3_Puya 
>  build_flags               = ${esp82xx_defaults.build_flags} -DARDUINO_ESP8266_RELEASE_2_4_1 
>   
>  [core_2_4_2] 
>  platform                  = https://github.com/TD-er/platform-espressif8266.git#patch/v1.8.0_Puya 
>  build_flags               = ${esp82xx_defaults.build_flags} -DARDUINO_ESP8266_RELEASE_2_4_2 
>   
>  [core_2_5_0] 
>  plataforma                   = espressif8266@2.0.1
>  build_flags                = $ {esp82xx_2_5_0.build_flags}
>   
>  [core_stage] 
>  plataforma                   = https://github.com/platformio/platform-espressif8266.git # feature / stage 
>  build_flags                = $ {esp82xx_2_5_0.build_flags}
>   
> Você pode ver as seções marcadas com algo parecido `[core_stage]`e `[esp82xx_2_5_0]`
> Como você pode ver, a `core_stage`seção se refere `esp82xx_2_5_0`no `build_flags`parâmetro.
> 
> Agora, o mesmo para a próxima parte importante deste arquivo platformio.ini:
> 
> [ESPEasy / platformio.ini](https://github.com/letscontrolit/ESPEasy/blob/147da97f24f28e1776c562b4eac5f081cc690388/platformio.ini#L154-L180)
> 
> Linhas 154 a 180 em [147da97](/letscontrolit/ESPEasy/commit/147da97f24f28e1776c562b4eac5f081cc690388)
> 
>  [debug_flags] 
[... 82 lines cut ...]
>  lib_archive                = $ {common.lib_archive}
>  framework                  = $ {common.framework}
>  upload_speed               = $ {common.upload_speed}
>  monitor_speed              = $ {common.monitor_speed}
>  board                      = $ {esp8266_1M.board}
>  board_upload.maximum_size = $ {esp8266_1M.board_upload.maximum_size}
>  board_build.f_cpu          = $ {esp8266_1M.board_build.f_cpu}
>  board_build.flash_mode     = $ {esp8266_1M.board_build.flash_mode}
>  build_unflags              = $ {esp8266_1M.build_unflags}
>  build_flags                = $ {esp8266_1M.build_flags} $ {normal.build_flags}
>  targets                    = $ {common.targets}
> Como você pode ver, estas são apenas referências a seções que fizemos anteriormente.
> 
> Mas a parte mais importante que você precisa saber é que tudo se resume a adicionar a bandeira PUYA ao `build_flags`. 
> Então, essa lista de flags terá muitos parâmetros e um deles é o `PUYA_SUPPORT=1`
> Flags definido na lista que deve estar `-D`na frente deles, então `-D PUYA_SUPPORT=1`ou `-DPUYA_SUPPORT=1`. 
> Então a `build_flags`linha será algo assim:
> 
> ```
> build_flags               = ....lots_of_other_flags...  -DPUYA_SUPPORT=1
> ```
> Espero que agora esteja claro como isso funciona.

It is not clear to me what I should do with this platformio, should I edit my or do douwnload and replace.
Where I find my platformio, I searched the filename and it has several platformio. The core I am using is 2.4.0

TD-er, 2019-03-21:
What are you trying to do?
Building ESPeasy?
Then just download it from the Github repository and there you have a platformio.ini file that simply works.
Or have at least a look at how the files are organized in a src directory, lib directory and a platformio.ini file.

In Arduino IDE the main .ino file (the one with the `loop()` and `setup()` function) must have the same name as the directory it is in (so `ESPeasy.ino` is in the directory `ESPeasy`)
PlatformIO is a bit more flexible, as long as you only have one .ino file in the src directory that has a `setup()` and `loop()` function.

This is becoming more a discussion on how to start using Arduino IDE or PlatformIO and not about ESPeasy.
Also how to use stuff in the core library is more a question for the core library repository.
I already spent several hours on explaining how to build the core library, while it is actually a question to be asked to the maintainers of the core library.

Just to give some extra links for you to continue:

- [https://github.com/esp8266/Arduino/issues](ESP8266 Arduino issues)
- [http://docs.platformio.org/en/latest/projectconf.html](How to use Platformio.ini file)
- [http://docs.platformio.org/en/latest/frameworks/arduino.html](How to use Arduino as a platform in PlatformIO)
- [http://docs.platformio.org/en/latest/ide/atom.html](How to use PlatformIO in Atom (editor))
- [http://docs.platformio.org/en/latest/faq.html](Platform IO FAQ)

adapso, 2019-03-21:
I'm trying to make my ESP PUYA work. I have tried several times without success in arduino IDE, there I was suggested trying to use plataform.IO, so I did this now do not know how to make the necessary changes inside it to work. I do not use ESPeasy.

TD-er, 2019-03-21:
Please read the links I placed in my post.
They should give enough information to get started in almost every topic you will probably encounter in this process.