letscontrolit/ESPEasy #751: Time and Date System Variables do not work in OLED display. Release v2.0-20180121

Budman1758, 2018-01-21:
Using and OLED display. (not framed) the following system variables do not display correctly.  All you get is exactly what you type.

%syshour%  $sysmin%  %syssec%  %sysday%  %sysmonth%  %sysyear%  %sysyears%  

All others listed on the system variables page work OK.  Don't know about the one for voltage.
[https://www.letscontrolit.com/wiki/index.php/ESPEasy_System_Variables](url)

These were working fine in the  Release v2.0-20180116 version.

As for the OLED framed display, is it broken?  All I get from is is garbage on the display.  Do I need to create an issue for that or is it a known issue?  Or is it just me?

TD-er, 2018-01-21:
It is a known issue and the solution will be code cleanup.
The problem is that lots of these system values are being programmed over and over again and sometimes a few of them are missing.

The voltage version is a special case. It has to be enabled during compile time and I am not sure why it is disabled. Probably for a good reason :)

psy0rz, 2018-01-21:
Related to #698

TD-er, 2018-01-21:
Also related to #709

TD-er, 2018-01-21:
@Budman1758 
While testing for the bugfix, I noticed you had a $ instead of a % in your testset :)

But it should now be equal for all plugins, rules and also for all controllers.
The system variables should be available to all, the sensor values only to the rules and plugins.

Budman1758, 2018-01-22:
Sorry for that.  Just a typo in the post.  Used % in all my tests.

closed 2018-01-22 by psy0rz