letscontrolit/ESPEasy #752: Cayenne-MQTT Support for ESPEasy

mstjerna, 2018-01-21:
Hi

I was speaking to "TD-er" on the forum regarding support for the Cayenne MQTT integration.
https://www.letscontrolit.com/forum/viewtopic.php?p=22775

The Cayenne uses longer MQTT username/clientID/passwords that currently can't be entered in the ESPEasy webpage. (Cayenne uses 36, 40, 36 char)

It would be a great extension to the ESPEasy if users could just create a dashboard for there sensor without the need for setting up there own OpenHAB or similar HomeAutomation server. 
Cayenne offers a nice dashboard on the web as well as Android/iOS mobile clients...

There are already open source examples from Cayenne in how to integrate with them here:
https://github.com/myDevicesIoT/Cayenne-MQTT-ESP

denisfrench, 2018-09-29:
Just to consolidate; Losant has similar issues with long-ish ControllerUser (36 char + NUL = 37) and ControllerPassword (64 char + NUL = 65) requirements. See PR #665.

Unfortunately (a separate issue) Losant will also immediately drop the connection if LWT message is sent with the retain flag set. See PR #925.

GeneArnold, 2019-07-17:
I wanted to follow up with this. It seems like these longer usernames and passwords are common. I had the same issue with QIoT and now with Cayenne. I've asked before about getting longer username and password options and have been told that it would be quite the task. I'm just posting here to say that it would be a nice enhancement. Thank you

bsimmo, 2019-12-07:
Before I start having ago with sending to Cayenne like I do with my Pi with my espeasy devices.
 A quick search for how to comes across this issue.

Has it been sorted (I'm using the latest 2.x builds) or should I look elsewhere? (it's still open but that doesn't always mean it is not solved)

TD-er, 2019-12-07:
The max length of passwords has not yet changed.
I did just a quick re-read of this issue, so that seems to be the only issue for this, right?

denisfrench, 2019-12-08:
Looks like all of username, clientID and password need to be longer for Cayenne (36, 40, 36 char per the first post, but I don't use it).

I do use Losant, which needs the username extended to 36 char, and password to 64.

Looks like this might be a bit easier to accommodate now with the new memory layout?

bsimmo, 2019-12-24:
Thanks for having a look at this

bsimmo, 2020-04-04:
How is this going as I've just grabbed the latest and I still cannot get it to connect to Cayenne ?
I'm assuming that's still passwords being the problem (that's all it says in DebugDev weblog)
```
2481261: MQTT : Intentional reconnect
2481475: MQTT : Broker C005 connection failed (3/0)
```

denisfrench, 2020-04-04:
@TD-er  is working on it now; see PR #2972 .

tonhuisman, 2023-09-02:
This has been resolved some time ago, so this issue can be closed.

closed 2023-09-02 by TD-er