letscontrolit/ESPEasy #753: CRC fail notification

s0170071, 2018-01-21:
as of https://github.com/letscontrolit/ESPEasy/pull/720 there are crc (md5 to be precise) on Settings, SecuritySettings and the SPI flash (program code). So far there are only warnings Serial about
 
- crc OK/fail of settings / securitysettings
- program code has changed since last save of settings (either) 
- crc OK/fail on program code
- crc not found in program code (someone forgot to run crc2.py or similar)

There should be some other consequences because not everyone is monitoring serial.
It should be displayed in e.g. the web interface, a faulty CRC on settings may e.g. lead to reduced functionality etc... 

Thread is open for discussion...

Budman1758, 2018-01-21:
Maybe a message box on the "Main" page where they would stick until the user can clear it?  Or see it for that matter. :)

TD-er, 2018-01-29:
The CRC checksums are only available on the Mega branch.
I have added a similar check/report on the use of MQTT controllers. (see #740 )
So perhaps you can check with the latest 2.0 sources (@psy0rz will merge to Mega later today I guess) if the error given when setting 2 MQTT controllers active is somewhat like you imagined for this check?
Then I can add a similar check for the CRC errors. (which are a bit harder to test :) )

TD-er, 2018-01-29:
It is harder to create an incorrect CRC on saved files.
[image]
As an example on how I changed it with the MQTT controllers. The message is then shown on all pages (alert box can be closed)

s0170071, 2018-01-29:
Nice. To trigger that box we could use a global string or so. Then the load / save settings functions can add text to that box. Possible strings are
- binary changed
- CRC fail.
If the string is empty, no box is shown.

TD-er, 2018-01-29:
Just a global check function. Maybe even allow some static files in that function, to only update it when files are actually read. Otherwise you would read the files on flash on every webpage change.

s0170071, 2018-01-29:
Forget that string. That's stupid. We also need the possibility to remove errors when they don't apply any more.
That static function needs to check an array of flags and assemble the text based on those flags. Each error has a flag associated to it. 
Not perfect but the best I can think of now.

closed 2019-10-11 by Grovkillen