letscontrolit/ESPEasy #755: Uptime message twice per minute ?

s0170071, 2018-01-21:
wasn't it only once per minute?

```
WIFI : Connected! IP: 192.168.1.14 (ESP_Easy_100)
WD   : Uptime 0 ConnectFailures 0 FreeMem 22584
WD   : Uptime 1 ConnectFailures 0 FreeMem 22416
WD   : Uptime 1 ConnectFailures 0 FreeMem 22416
WD   : Uptime 2 ConnectFailures 0 FreeMem 22416
WD   : Uptime 2 ConnectFailures 0 FreeMem 22416
WD   : Uptime 3 ConnectFailures 0 FreeMem 22416

```

Software or git version:  mega, 2.3.0

Budman1758, 2018-01-21:
I think it was only once a minute.  I noticed it was twice a minute but didnt think too much of it.

Grovkillen, 2018-01-21:
It has "always" been that way. I figured it was to make the life ping wait shorter.

s0170071, 2018-01-21:
Well then...

closed 2018-01-21 by s0170071