letscontrolit/ESPEasy #759: P052 SenseAir ABC calibration feature WARNING

Grovkillen, 2018-01-22:
https://github.com/letscontrolit/ESPEasy/pull/606/files

We need to add information that ABC period change from X to X-y (i.e. shorter period) will result in a time out error since we have a bug in the current S8 firmware. The ABC period should only be changed right after a ABC is done, at least if you want to shorten the ABC period.

A notification in the web form should be enough I think (I will add this information to the wiki as well).

TD-er, 2018-01-22:
Is there an option to see when the ABC in being performed?

Does this bug also appear on other sensors?

Grovkillen, 2018-01-22:
Only the S8 I've been told. I think it's possible to get the information that a ABC has been done. But I dont think this is stored as a value only a signal among signals so to speak. Maybe setting the ABC to disabled and then enable it after a short while is resetting the timer (don't know that for sure).

https://rmtplusstoragesenseair.blob.core.windows.net/docs/Dev/publicerat/TDE2067.pdf

TD-er, 2018-01-22:
Does it remain giving timeouts, or is a power cycle enough to get it working again?

Grovkillen, 2018-01-23:
I'm going to get info on that one but I think a power cycle is not helping.

TD-er, 2018-01-23:
Is re-setting to the previous value enough?
Can you always go back to the factory default value, or is increasing the value the only option?
Should we disable the option for now?
Can we add a firmware upgrade as an option?
Did I miss any other concern? ;)

Grovkillen, 2018-01-23:
I've been told that resetting the value back to what it was is not a solid fix since the problem is that the timer used between ABC is triggered by a "TimerNOW=TimerABCperiod" and not  "TimerNOW>=TimerABCperiod" plus that the timer is not set to zero upon change of ABC period.

So the only way of manually resetting the ABC is to do it right after a ABC is done.

Upgrade of FW is possible but not in the scoop right now, lets disable the option and let advanced users compile their own (or if it is available in dev version?) FW.

Also add the information as comments in the source code would be nice, even experienced users may not understand this bug.

TD-er, 2018-01-23:
OK, then we will change it.
I will also warn Theo Arends (TASMOTA) about this, since he has added support for the S8.

TD-er, 2018-01-23:
Hmm.. isn't there a command to get the time until the next ABC? Or was that not in the S8 sensor?
I have read way too many datasheets in the last months, things start to mix up in my head.

Grovkillen, 2018-01-23:
I don't think there is a command for fetching that. Not in the S8.

Great, notice TASMOTA community as well. Thanks!

TD-er, 2018-01-24:
Set into comments.

I think this plugin deserves a rewrite, since it is quite different from other plugins.
You can read other values, but have to choose.
And also I really like to have a central Modbus interface to be able to use with other devices.
So expect some major rewrite in the near future :)

Grovkillen, 2018-01-24:
Looking forward to it :)

Grovkillen, 2018-01-27:
Thanks for this @td-er

closed 2018-01-27 by Grovkillen