letscontrolit/ESPEasy #760: Support for an Dustsensor PMS5003

micropet, 2018-01-22:
Hi,

I want use  a Dustsensor PMS5003 or PMS7003 in ESPEasy.
Is support planned for this?

Greetings Peter

TD-er, 2018-01-22:
The SDS011 is already supported.
I don't know the PMS ones by experience.
What features make the PMS ones more interesting than the SDS011?

I'll have to order them to get it supported.

micropet, 2018-01-22:
Cheap and very small
Able to sense PM10 and PM2.5 particles !!
No need to calibrate, and reasonable calibration from the factory.

TD-er, 2018-01-22:
Hmm, the first few hits I see are not what I call "cheap".
More in the price range "Not an option for me".
So do you have a link with more information and a reasonable price?

The SDS011 (about US$ 20) can do PM10 and PM2.5 and also do not require calibration.
The measured values I get correspond quite well with the official readings from the weather institute in the Netherlands.

micropet, 2018-01-22:
Ok. 23 §
https://de.aliexpress.com/item/PLANTOWER-Laser-PM2-5-DUST-SENSOR-PMS7003-High-precision-laser-dust-concentration-sensor-digital-dust-particles/32623909733.html

You're right. I thought that the SDS011 can only PM2.5.
I have to compare the sensors in peace.
Thank you.

TD-er, 2018-01-22:
23$ is not too bad for a sensor. 
I looked at the PMS5003 and that one was about 140-ish $ and about as big as a book.

ShardanX, 2018-01-22:
Have a look at the wiki, there are some photos showing the PMS5003 and the even smaller 7003, i've tested both with ESPEasy and they work fine. Calibration is not required, the PMS x003 can gauge PM1.0, PM2.5 and PM10.

I'm at the office atm and can't take a look, it might be possible that you have to install the TESTING version. The PMS7003 is running for several weeks meanwhile on my test site.

The Wiki describes the setup including a workaround for a bug that limits the lifetime of the sensor's laser.
https://letscontrolit.com/wiki/index.php/PMSx003.

Regards
Shardan

Grovkillen, 2018-01-22:
I should close this, right @ShardanX??

micropet, 2018-01-23:
Thank you Shardan,
I will test it the days.

Yesterday I put a Wemos D1 with BME280, CO2 sensor MH-Z19 and Dustsensor SDS021 in operation.
The values of the SDS look plausible.
The MH-Z19 sometimes causes problems and can not be read.

Are two sensors on softserial perhaps already too much?

Greetings Peter

TD-er, 2018-01-23:
Did you put both sensors on separate GPIO pins?
The SDS011 can operate with just 1 wire (sensor TX to RX on ESP board), with as limitation that it is not possible to change the interval setting (time on, time off to save the laser's lifetime)
That value is stored in the sensor, so setting it can be done once and then you can re-use that pin on the ESP.

The MHZ19 and the SDS011 both consume quite a lot, so please also have a look at the power supply, the cables to the sensors, etc.

But I guess this is more of an issue to discuss on the forum, unless you really found something that is wrong in the code.

micropet, 2018-01-23:
Yes, on separate pins:

D3 GPIO0 RX from the MH-Z19 orange 2.GPIO with software Serial
D4 GPIO2 TX from MH-Z19 o / sw 1.GPIO with Software Serial

D7 GPIO13 TX from SDS021 blue with software serial
D8 GPIO15 RX from SDS021 yellow with software serial

Sometimes a CO2 value is also displayed. Usually comes:
MHZ19: Error, timeout while trying to read
MHZ19: Unknown response: 0 0 0 0 0 0 0 0 0

I check the 5V supply later.

I've just signed up for the forum and I'm still waiting for the confirmation.
Thanks for the answer.

TD-er, 2018-01-23:
Software Serial does support multiple concurrent ports.
In my last change, I reduced the number of concurrent ports to 3 (often limited by the number of free GPIO pins) to reduce memory use of the very limited available iRAM cache.

micropet, 2018-01-23:
OK, then it should work.
I'll check the 5V later.
Or I use other pins.

The MH-Z19 worked fine until yesterday with an ESP32 board.

Since ESPEasy does not yet support the ESP32, I built a Wemos D1 into the case.

Grovkillen, 2018-01-23:
I close this now. You may continue the conversation but the issue is closed.

closed 2018-01-23 by Grovkillen