letscontrolit/ESPEasy #761: Feature request: control buttons for actuators

s0170071, 2018-01-22:
that's a feature that I like about tasmota... Big on/off buttons on the main page. For the mobile phone users.
[image]

That made me think... 
What if we had a plug in that can configure maybe four of these buttons ? Or sliders or anything else that allows actuators to be controlled ? Or can we do that already and I do not know ?
Anyway, the main page of ESPEasy could offer a bit more everyday functionality. Sensor reading, maybe a gauge, a button. 
The node list, for example. Most ESPs have only one sensor attached, so why not display one or two readings in that list. You open one device and can have a glance at them all. 
Don't get me wrong, please, but the main page could use some more of that "user experience" thing ;-)

TD-er, 2018-01-22:
I totally agree.
We were already thinking of adding a more "REST" like approach to ESPeasy.
This allows for a lot of nice new features:

- More interactive web frontend.
- Inclusion in other portal sites (central site to operate all ESPeasy nodes)
- Easy backup of settings
- etc.

When such an interface is present, anyone can make all sorts of interfaces and even combine plugins in a simple view.
The template and static files for the webpage could then be stored in SPIFFS (as long as it isn't a PUYA chip ;) ), etc.

s0170071, 2018-01-22:
you can store it in PUYA but you can't read it ;-) It makes it a  **P**otentially **U**nusable **Y**ield **A**pplication which **P**robably **U**nsatisfies **Y**ou **A**nyway.

When you write easy backup, I remember I recently thought about storing the settings in XML. Or at least make them accessible (read/write, not necessarily store) as XML. That would make commissioning a series of sensors quite easy. You just copy the settings, edit them with an editor and thats it. 

Even without XML we would also store them a bit more flexibly. I was thinking of a small struct for each value, stating the current size, max size, ID and the value. Maybe the unit and the name too. This way you could modify value sizes in the Settings struct without ruining everybody’s settings. I saw a PR about sth similar. 
When reading the settings you would rather parse the file than just map it onto the struct.

TD-er, 2018-01-22:
@s0170071 we (@Grovkillen and I) discussed this already and I am already looking into it.