letscontrolit/ESPEasy #763: Error compiling e36a9f163e26fbd42e722dd78f33c78a76c1103e

micropet, 2018-01-23:
Hello everybody,
I use the Arduino IDE 1.86 and the latest esp8266 sources.

When compiling the current mega  version for a Wemos D1, I get the following error message.

In file included from D:\arduino-1.8.5\hardware\esp8266com\esp8266\cores\esp8266/Arduino.h:241:0,

                 from sketch\ESPEasy.ino.cpp:1:

D:\arduino-1.8.3\meine\ESPEasy-mega\ESPEasy\Misc.ino: In function 'void dump(uint32_t)':

D:\arduino-1.8.5\hardware\esp8266com\esp8266\cores\esp8266/pgmspace.h:115:55: error: invalid conversion from 'uint32_t {aka unsigned int}' to 'const void*' [-fpermissive]

 #define pgm_read_byte(addr) pgm_read_byte_inlined(addr)

                                                       ^

D:\arduino-1.8.3\meine\ESPEasy-mega\ESPEasy\Misc.ino:752:28: note: in expansion of macro 'pgm_read_byte'

           Serial.print  (  pgm_read_byte(a),HEX);

                            ^

D:\arduino-1.8.5\hardware\esp8266com\esp8266\cores\esp8266/pgmspace.h:101:23: error:   initializing argument 1 of 'uint8_t pgm_read_byte_inlined(const void*)' [-fpermissive]

 static inline uint8_t pgm_read_byte_inlined(const void* addr) {

                       ^

exit status 1
Fehler beim Kompilieren für das Board WeMos D1 R2 & mini.

Lstt2005, 2018-01-23:
Same problem...

TD-er, 2018-01-23:
What version of ESP8266 library do you use?
We currently only support 2.3.0.
The latest version is 2.4.0

micropet, 2018-01-23:
There you are again. .)

I use the current version from
https://github.com/esp8266/Arduino/commits/master
So 2.4.0

The mega version from yesterday compiles and runs.

psy0rz, 2018-01-23:
i've added #720 last night, it seems that it has code that doesnt compile under arduino 2.4.0. we'll need to look into it.

Lstt2005, 2018-01-23:
`D:/!0!!/ESPEasy/src/Misc.ino: In function 'void dump(uint32_t)':
C:\users\user\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/pgmspace.h:115:55: error: invalid conversion from 'uint32_t {aka unsigned int}' to 'const void*' 
[-fpermissive]
#define pgm_read_byte(addr) pgm_read_byte_inlined(addr)
^
D:/!0!!/ESPEasy/src/Misc.ino:1503:26: note: in expansion of macro 'pgm_read_byte'

{
^
C:\users\user\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/pgmspace.h:101:23: error:   initializing argument 1 of 'uint8_t pgm_read_byte_inlined(const void*
)' [-fpermissive]
static inline uint8_t pgm_read_byte_inlined(const void* addr) {
^
D:/!0!!/ESPEasy/src/Misc.ino: In function 'uint32_t progMemMD5check()':
D:/!0!!/ESPEasy/src/Misc.ino:1549:63: warning: pointer of type 'void *' used in arithmetic [-Wpointer-arith]
{
^
C:\users\user\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/pgmspace.h:123:68: note: in definition of macro 'pgm_read_dword'
#define pgm_read_dword(addr)   (*reinterpret_cast<const uint32_t*>(addr))
^
Compiling .pioenvs\normal_ESP8266_4096\FrameworkArduino\StreamString.o
Compiling .pioenvs\normal_ESP8266_4096\FrameworkArduino\Tone.o
Compiling .pioenvs\normal_ESP8266_4096\FrameworkArduino\Updater.o
Compiling .pioenvs\normal_ESP8266_4096\FrameworkArduino\WMath.o
Compiling .pioenvs\normal_ESP8266_4096\FrameworkArduino\WString.o
Compiling .pioenvs\normal_ESP8266_4096\FrameworkArduino\abi.o
*** [.pioenvs\normal_ESP8266_4096\src\ESPEasy.ino.o] Error 1
 [ERROR] Took 5.62 seconds
 
 [SUMMARY]
Environment normal_ESP8266_1024 [SKIP]
Environment normal_ESP8285_1024 [SKIP]

`

closed 2018-01-25 by micropet