letscontrolit/ESPEasy #768: Cannot connect

s0170071, 2018-01-23:
just updated to the lates mega and entered my ssid / pass into the default section of ESPEasy.ino
boot...
```

INIT : Booting version: (custom)
INIT : Warm boot #5
FS   : Mounting...
FS   : Mount successful, used 76053 bytes of 957314
CRC  : program checksum       ...OK
CRC  : Settings CRC           ...FAIL
CRC  : SecuritySettings CRC   ...FAIL
INIT : Free RAM:23208
INIT : I2C
INIT : SPI not enabled
WIFI : AP Mode disabled
WIFI : AP Mode ssid will be ESP_Easy_11 with address 192.168.4.1
WIFI : Could not connect to AP!
WIFI : AP Mode enabled
WD   : Uptime 0 ConnectFailures 0 FreeMem 18832
WD   : Uptime 1 ConnectFailures 0 FreeMem 18936
WD   : Uptime 1 ConnectFailures 0 FreeMem 18936

```

doesn't connect.

TD-er, 2018-01-23:
Settings CRC... Perhaps the settings are corrupted now?

s0170071, 2018-01-23:
They never had the CRC yet, so yes. But what about the default values, shouldn't they apply ?

TD-er, 2018-01-23:
The CRC patch was committed yesterday

s0170071, 2018-01-23:
I know. Nice to see it working. Still, what about the default values, shouldn't they apply ? I mean even if there is no config. I believe that always worked so far. Until now.

TD-er, 2018-01-23:
Were other settings still working?

I haven't tried it yet, but I can imagine the CRC was not yet present on the existing config files.

s0170071, 2018-01-23:
The other settings are ok, and the CRC was definitely not present. 
Under which circumstances does the default value apply ?

TD-er, 2018-01-23:
If I remember correctly, the default values are only used when the device is set to "factory default", like when you start with an empty configuration.

But if you had previous settings, those values should still be valid.
The reported CRC error is only that, being reported. It is not being used as rejection.

psy0rz, 2018-01-23:
yeah default values are only applied to ALL settings on a full reset. (it will start by formatting spiffs)

s0170071, 2018-01-27:
oh. Well, closing then. 
Still. if no connection is available (or crc is corrupt ) using the default would be an option. 
I rather reflash the ESP than adjust the wifi with using my smartphone...

s0170071, 2018-01-27:
:-)
Github says: s0170071 closed this an hour from now... wtf ?

closed 2018-01-27 by s0170071