letscontrolit/ESPEasy #769: Error compiling mega mega-20180124

Lstt2005, 2018-01-24:
Hello, when i try compiling from source in PlatfprmIO, i get errors:
Project
ESPEasy
.github
dist
lib
src
__CPlugin.ino
__NPlugin.ino
__Plugin.ino
__ReleaseNotes.ino
_C001.ino
_C002.ino
_C003.ino
_C004.ino
_C005.ino
_C006.ino
_C007.ino
_C008.ino
_C009.ino
_C010.ino
_C011.ino
_N001_Email.ino
_N002_Buzzer.ino
_P001_Switch.ino
_P002_ADC.ino
_P003_Pulse.ino
_P004_Dallas.ino
_P005_DHT.ino
_P006_BMP085.ino
_P007_PCF8591.ino
_P008_RFID.ino
_P009_MCP.ino
_P010_BH1750.ino
_P011_PME.ino
_P012_LCD.ino
_P013_HCSR04.ino
_P014_SI7021.ino
_P015_TSL2561.ino
_P016_IR.ino
_P017_PN532.ino
_P018_Dust.ino
_P019_PCF8574.ino
_P020_Ser2Net.ino
_P021_Level.ino
_P022_PCA9685.ino
_P023_OLED.ino
_P024_MLX90614.ino
_P025_ADS1115.ino
_P026_Sysinfo.ino
_P027_INA219.ino
_P028_BME280.ino
_P029_Output.ino
_P030_BMP280.ino
_P031_SHT1X.ino
_P032_MS5611.ino
_P033_Dummy.ino
_P034_DHT12.ino
_P035_IRTX.ino
_P036_FrameOLED.ino
_P037_MQTTImport.ino
_P038_NeoPixel.ino
_P039_Thermocouple.ino
_P040_ID12.ino
_P041_NeoClock.ino
_P042_Candle.ino
_P043_ClkOutput.ino
_P044_P1WifiGateway.ino
_P045_MPU6050.ino
_P046_VentusW266.ino
_P047_i2c-soil-moisture-sensor.ino
_P048_Motorshield_v2.ino
_P049_MHZ19.ino
_P050_TCS34725.ino
_P051_AM2320.ino
_P052_SenseAir.ino
_P053_PMSx003.ino
_P054_DMX512.ino
_P055_Chiming.ino
_P056_SDS011-Dust.ino
_P057_HT16K33_LED.ino
_P058_HT16K33_KeyPad.ino
_P059_Encoder.ino
_P060_MCP3221.ino
_P061_KeyPad.ino
_P062_MPR121_KeyPad.ino
_P063_TTP229_KeyPad.ino
_P064_APDS9960.ino
_P065_DRF0299_MP3.ino
_P066_VEML6040.ino
_P067_HX711_Load_Cell.ino
_P068_SHT3x.ino
_P069_LM75A.ino
_P070_NeoPixel_Clock.ino
_P071_Kamstrup401.ino
_P072_HDC1080.ino
_P073_7DGT.ino
_P074_TSL2591.ino
_Reporting.ino
Command.ino
Controller.ino
Dialog_Plain_12_font.h
ESPEasy.ino
ESPEasyTimeTypes.h
Hardware.ino
Misc.ino
Networking.ino
OLED_SSD1306_images.h
OLED_SSD1306_SH1106_images.h
Serial.ino
TimeESPeasy.ino
TimeZoneESPeasy.ino
WebServer.ino
Wifi.ino
test
.atom-build.yml
.gitignore
.gitmodules
.travis.yml
before_deploy
License.txt
memanalyzer.py
platformio.ini
README.md
release
releasebot
Welcome
A hackable text editor for the 21st Century

For help, please visit

The Atom docs for Guides and the API reference.
The Atom forum at discuss.atom.io
The Atom org. This is where all GitHub-created Atom packages can be found.
Show Welcome Guide when opening Atom
atom.io × 
Welcome Guide
PlatformIO Home
ESPEasy.ino

9.4 s
platformio run --environment dev_ESP8266_4096
static inline uint8_t pgm_read_byte_inlined(const void* addr) {
^
D:/!Git/ESPEasy/src/Misc.ino: In function 'uint32_t progMemMD5check()':
D:/!Git/ESPEasy/src/Misc.ino:1549:63: warning: pointer of type 'void *' used in arithmetic [-Wpointer-arith]
{
^
C:\users\user\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/pgmspace.h:123:68: note: in definition of macro 'pgm_read_dword'

#define pgm_read_dword(addr)   (*reinterpret_cast<const uint32_t*>(addr))
^
Compiling .pioenvs\dev_ESP8266_4096\FrameworkArduino\Tone.o
Compiling .pioenvs\dev_ESP8266_4096\FrameworkArduino\Updater.o
Compiling .pioenvs\dev_ESP8266_4096\FrameworkArduino\WMath.o
Compiling .pioenvs\dev_ESP8266_4096\FrameworkArduino\WString.o
Compiling .pioenvs\dev_ESP8266_4096\FrameworkArduino\abi.o
Compiling .pioenvs\dev_ESP8266_4096\FrameworkArduino\base64.o
*** [.pioenvs\dev_ESP8266_4096\src\ESPEasy.ino.o] Error 1
 [ERROR] Took 8.45 seconds
Environment dev_ESP8266_4096    [ERROR]
 [ERROR] Took 8.46 seconds
 
 [SUMMARY]
Environment normal_ESP8266_1024 [SKIP]
Environment normal_ESP8266_4096 [SKIP]
Environment normal_ESP8285_1024 [SKIP]
Environment test_ESP8266_1024   [SKIP]
Environment test_ESP8266_4096   [SKIP]
Environment test_ESP8285_1024   [SKIP]
Environment dev_ESP8266_1024    [SKIP]
Environment dev_ESP8285_1024    [SKIP]

What is my mistake?

TD-er, 2018-01-24:
What version of the esp8266 library do you use?
We only support 2.3.0 currently

Lstt2005, 2018-01-24:
I use 2.4.0 version....

Lstt2005, 2018-01-24:
Can you help me instal version 2.3.0 correct?

TD-er, 2018-01-25:
@Lstt2005 Suggestion from another thread:
https://github.com/letscontrolit/ESPEasy/pull/772#issuecomment-360446929

> You have to edit the platformio.ini file!
> change the following key for each build:
> platform = espressif8266 =>platform = espressif8266@1.5.0
> 
> You now force platform I/O to use a specific version. Normally it uses the staging version from github

Lstt2005, 2018-01-25:
All Work! Thanks for all!

closed 2018-01-25 by Lstt2005