letscontrolit/ESPEasy #770: Target rebooting if Plugin Communication - P1 Wifi Gateway is selected, Release v2.0-20180124

Rewotek, 2018-01-24:
Hi All

Precompiled binary (ESP_Easy_v2.0-20180124_normal_ESP8266_1024.bin) loaded on ESP-07, when P1 Wifi Gateway is selected from Devices target is rebooting with the following serial log (partial):
Earlier release (20180114 for instance) seems not affected.

Data received: 
Panic /home/travis/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/core_esp8266_main.cpp:98 __yield

ctx: cont 
sp: 3fff3590 end: 40000f83 offset: 01b0

>>>stack>>>
3fff3740:  3fff14e2 3fff15ec 40203464 3fff28e0  
3fff3750:  00000000 3fff6b14 00000000 402034ed  
3fff3760:  3fff14e2 00000000 3fff08fc 40233445  
3fff3770:  00000000 3fff15ec 00047c41 40233474  
3fff3780:  3fff14e2 3fff15ec 3fff37d0 40233515  
3fff3790:  00000000 00000000 00000000 40202536  
3fff37a0:  3ffe870f 00000004 3fff37d0 40100688  
3fff37b0:  00000000 00000000 3fff37d0 3fff15ec  
3fff37c0:  3fff14e2 3fff0b74 00000010 4023373d  
3fff37d0:  00000100 00000400 00000000 40202401  
3fff37e0:  00000000 00000000 00000000 00000000  
3fff37f0:  00000000 00000100 3fff5154 0000000f  
3fff3800:  00000000 3fff6a3c 0000000f 00000000  
3fff3810:  3fff6a54 0000000f 00000000 00000000  
3fff3820:  40800000 00000000 00000000 00000000  
3fff3830:  3fff0934 00000005 00000001 4021c07d  
3fff3840:  00000600 00001800 00000000 0000000a  
3fff3850:  00000000 00000000 00000000 00000000  
3fff3860:  00000000 3a643906 00045b81 40215d0f  
3fff3870:  00000000 00000000 00000001 00000000  
3fff3880:  3fff19a0 3fff0b18 00000001 402337bc  
3fff3890:  3fff5e8c 4020d064 00000006 3fff28d4  
3fff38a0:  3fffdad0 00000000 3fff091c 402377ff  
3fff38b0:  00000000 00000000 000443d8 40215d0f  
3fff38c0:  3fffdad0 00000000 3fff0930 3fff28d4  
3fff38d0:  3fffdad0 00000000 
Data received: 05f08  
3fff3900:  feefeffe 3fff28e0 00000003 00000004  
3fff3910:  40203c44 00000000 00000000 00000000  
3fff3920:  00000000 00000001 3ffe9dba 00000062  
3fff3930:  3ffe9d3e 00000000 4010750c 00000000  
3fff3940:  0e51fcf6 00000000 05979147 011e1a30  
3fff3950:  40203d50 00000000 ffffffff 058eb1f7  
3fff3960:  00000000 40203d38 00000000 00000000  
3fff3970:  00000000 00000000 00000000 00000000  
3fff3980:  00000000 00000000 00000000 00000000  
3fff3990:  00000000 00000000 00000000 00000000  
3fff39a0:  00000000 00000000 00000000 00000000  
3fff39b0:  00000000 00000000 00000000 00000000  
3fff39c0:  00000000 00000000 00000000 00000000  
3fff39d0:  00000000 00000000 00000000 00000000  
3fff39e0:  00000000 00000000 00000000 00000000  
3fff39f0:  03fe0001 00000000 00000000 00000000  
3fff3a00:  00000000 00000000 00000000 00000000  
3fff3a10:  00000000 00000004 00000000 00000000  
3fff3a20:  00000000 00000000 00000000 00000000  
3fff3a30:  00000000 00000000 00000000 00000000  
3fff3a40:  00000000 00000000 00000000 00000000  
3fff3a50:  00000000 00000000 00010101 0000004e  
3fff3a60:  00000000 00000001 00000000 00000000  
3fff3a70:  00000000 00000000 00000000 00000000  
3fff3a80:  00000000 00000004 00000000 3fff4f24  
3fff3a90:  3fff4fc4 ffffffff 00000ef4 3fff4860  
3fff3aa0:  00000000 00000000 00000000 00000000  
3fff3ab0:  00000000 0000000
Data received: 105e50a  
3fff3ae0:  00000000 00010000 00000000 00000000  
3fff3af0:  6c6f6f70 70746e2e 67726f2e 00000000  
[... 683 lines cut ...]
3fff6750:  4023f33c 00000000 00000000 00000000  
3fff6760:  00000000 3fff677c 0000000f 00000006  
3fff6770:  00000000 00000000 03db03e6 6c75722f  
3fff6780:  00007365 00000000 00000000 00000000  
3fff6790:  03e303ee 3ffea670 3fff67ec 40220298  
3fff67a0:  00000000 4023f350 4023f33c 00000000  
3fff67b0:  00000000 00000000 00000000 3fff67d4  
3æ
 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09826c6d
~ld

Data received: ªU

INIT : Booting version: v2.0-20180124
INIT : Warm boot #12
FS   : Mounting...
FS   : Mount successful, used 78563 bytes of 113201
[serial.log](https://github.com/letscontrolit/ESPEasy/files/1659230/serial.log)

TD-er, 2018-01-24:
The watchdog is kicking in.
I've seen the same with the Kamstrup, a few weeks ago.
I will look into it. (or @psy0rz if he realizes he did something last night that might be related ;) )

My first impression is that the plugin is trying to get data, but will timeout without proper settings.
If I'm correct, there is a command to delete the plugin via an URL.
Maybe @Grovkillen or @psy0rz know it by hart?

Grovkillen, 2018-01-24:
yeah you can do a taskclear via this command:

http://**<espeasyip>**/control?cmd=TaskClear,**<task/device nr>**

closed 2018-01-24 by Grovkillen