letscontrolit/ESPEasy #777: [2.4.0] ERR_INVALID_CHUNKED_ENCODING when saving settings on OLED framed

TD-er, 2018-01-25:
With the 2.4.0 core compiled version of the Mega branch.
When saving settings in the OLED Framed  plugin, Chrome reports ERR_INVALID_CHUNKED_ENCODING and refuses to continue.

Not sure yet what causes it.

s0170071, 2018-01-27:
I left a sketch a while ago here: https://github.com/esp8266/Arduino/issues/3871
It just returns an 8kb dummy website. Can you check if it also produces the chunking error  ? 
If it does, thats a core issue.

TD-er, 2018-01-27:
I notice you use a delay of 20 msec per request.
Others also mentioned something similar, which I do not understand why it should make a difference.
Is it possible requests are queued to be processed later and then the strings containing the page are no longer valid?

I will try your sketch later this evening.

Can I detect the LWIP version on runtime?

s0170071, 2018-01-27:
There are #defines for lwip but that's compile time. Why would you need run-time ? You can't switch it at run time. 
In case compile time does work for you, the defines are here: 
https://github.com/letscontrolit/ESPEasy/pull/707

As for the delay, that seemed to be some sort of race condition which hit me when I was connecting through VPN. Seems fixed though and should now work without.

TD-er, 2018-01-27:
It was just to check what version is actually used.
My experience with the libraries included in PlatformIO is that change in settings and force recompile is no guarantee it is actually being used.
So runtime check would be nice.

s0170071, 2018-01-27:
There is a lwip version include. Maybe pimp that one...
-- 
Diese Nachricht wurde von meinem Android-Gerät mit K-9 Mail gesendet.

s0170071, 2018-01-27:
Thinking of it... Can we show the lwip version on the info page? I got the feeling that question will come up more often now...

TD-er, 2018-01-27:
That last one was exactly what I was thinking about.
Showing lib versions used.

s0170071, 2018-01-27:
For that purpose a #define would suffice. 
If platformio does not re-build the lwip if you changed the settings, that would also mean that it does not read the new #define. ( because object file is already there, it is skipping the source...)
I have difficulties imagining that it processes a source file and does not update the object file that belongs to it. Maybe if it cannot write the obj file to disk...

s0170071, 2018-01-28:
When I submit the framed oled settings, I first got a garbled webpage. After reset it works better. 
Still Firefox says:

> A form was submitted in the windows-1252 encoding which cannot encode all Unicode characters, so user input may get corrupted. To avoid this problem, the page should be changed so that the form is submitted in the UTF-8 encoding either by changing the encoding of the page itself to UTF-8 or by specifying accept-charset=utf-8 on the form element.
> devices
> The character encoding of the HTML document was not declared. The document will render with garbled text in some browser configurations if the document contains characters from outside the US-ASCII range. The character encoding of the page must be declared in the document or in the transfer protocol.
> devices

s0170071, 2018-01-28:
chrome does not give any errors, just favicon not found. Mega / 2.4.0 / Linux mint.

TD-er, 2018-01-28:
It would be nice if those ext. libraries already included such a define. I will look into it.

TD-er, 2018-01-28:
I've looked with Wireshark and it appears there is a difference at package level between save with OLEDframed and other devices.
There is an extra package with [RST, ACK] flags set.
It looks like the package is sent from my browser, not initiated from the ESP.

So I really have to dig deeper, since I have no clue what is going on here.
Firefox doesn't seem to have any issues here.

s0170071, 2018-01-28:
Is that a permanent issue or does it only occur occasionally ?
Using chrome, how do you detect the error ? Does it show in the developer tools / console ? Just asking cause I can't see it yet.

TD-er, 2018-01-28:
It is a permanent occurring issue. (using Chrome in Windows)
First a packet with [FIN, ACK] set and then one with [RST, ACK]

I've also looked into the OLED (non framed) version, since that one uses almost the same settings and then only the [FIN, ACK] packet is sent.

When using Chrome on Android, this does not occur.
One strange thing is that I only see packets with the ESP device as destination, not as src.
I can even see the DHCP, NTP and DNS queries and MQTT messages between ESP and Domoticz server, but Wireshark only shows the IP-address of the device in the dst column.
I set the capture filtering on "ether host a0:20:a6:19:2a:bf".

N.B. 
I use a so called "smart switch", which allows me to apply a port mirror of all traffic to a specific ethernet port on the switch.
This way I can see all traffic in my entire network using Wireshark. (all except traffic between devices on the same WiFi accesspoint, so for my tests I use different accesspoints.

s0170071, 2018-01-29:
Can I see it then without Wireshark in Chrome@Win7 ? If yes, where and whats the error message ?

TD-er, 2018-01-29:
I've also tested with Chrome-OS on a Chromebook and there the error is not displayed, but the page remains white.
The source of the page only contains text up to the end of the <meta> tag, nothing more.

In Chrome@Win10, it displays the following error:

> This site can’t be reached
> The web page at http://192.168.1.75/devices?index=2&page=1 might be temporarily down or it may have moved permanently to a new web address.
> ERR_INVALID_CHUNKED_ENCODING

s0170071, 2018-01-29:
I do experience (Linux / Win7  + Chrome / Firefox)  occasional lags of about 10s on various pages. But nothing like the chunking error.
On wednesday I get to try it on a win10 machine... 
The core devs would now ask if you have tried lwip2...  have you ?

TD-er, 2018-01-29:
Not as far as I know. But I haven't looked into the libs included in the PlatformIO ESP8266 1.6 library.

closed 2019-08-26 by TD-er