letscontrolit/ESPEasy #778: #define PLUGIN_BUILD_TESTING returns error messages

micropet, 2018-01-25:
Hello,

core 2.4.0 compiles well with #define PLUGIN_BUILD_DEV.

but #define PLUGIN_BUILD_TESTING returns error messages in misc.ino.

Version is b811b13483486cc423494a9f523bcf81b1671cd3

TD-er, 2018-01-25:
Can you please test with the latest source from Github?
I just compiled testing, without any error/warning.

Please also check the installed version of the PlatformIO esp8266 plugin.
Don't forget to restart PlatformIO, when you change branches. PlatformIO often ignores file changes done externally by Git.

If you still get errors, please include the errors.

micropet, 2018-01-25:
I have the latest source und use Arduino 1.8.5, here are the errors:

D:\arduino-1.8.5 - 2.4.0\libraries\ArduinoJson\src/ArduinoJson/Polyfills/./ctype.hpp:10:21: error: expected unqualified-id before 'char'

 inline bool isdigit(char c) {

                     ^

D:\TMP\arduino_modified_sketch_797968\Misc.ino:2455:21: note: in definition of macro 'isdigit'

 #define isdigit(n) (n >= '0' && n <= '9')

                     ^

D:\arduino-1.8.5 - 2.4.0\libraries\ArduinoJson\src/ArduinoJson/Polyfills/./ctype.hpp:10:21: error: expected ')' before 'char'

 inline bool isdigit(char c) {

                     ^

D:\TMP\arduino_modified_sketch_797968\Misc.ino:2455:21: note: in definition of macro 'isdigit'

 #define isdigit(n) (n >= '0' && n <= '9')

                     ^

exit status 1
Fehler beim Kompilieren für das Board WeMos D1 R2 & mini.

TD-er, 2018-01-25:
Looks like it is also part of the 2.4.0 code already. Which is strange, since isDigit was already part of the Arduino code. (differs only in capitalization)

jjansen85, 2018-01-26:
Had the same error last week (on 2.3.0). Apperently my ArduinoJson library was not up to date with the latest source code. I replaced with the one included in ESPEasy and it worked correctly

micropet, 2018-01-26:
I checked.
My ArduinoJson library is the one from ESPEasy Source.

Grovkillen, 2019-10-10:
I close this, open if still something needed.

closed 2019-10-10 by Grovkillen