letscontrolit/ESPEasy #780: Fatal exception error with ESP_Easy_v2.0-20180126_normal_ESP8266_4096

yoshibatora, 2018-01-28:
**NOTE: This is not a support forum! For questions and support go here: https://www.letscontrolit.com/forum/viewforum.php?f=1**

### Steps to reproduce
How can we trigger this problem?
I used a NodeMCU V3 like ESP8266 and flashed ESP_Easy_v2.0-20180126_normal_ESP8266_4096.bin, but debug found Fatal exception error. The detailed ESP8266 links below.
[http://developer.kaifakuai.com/product4.html](url)

Does the problem presist after powering off and on? (just resetting isnt enough sometimes)

### Expected behavior
Tell us what should happen?
[image]

### Actual behavior
Tell us what happens instead?
Debug shows below and no AP found.
epc1=0x400ef424, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Fatal exception (0): 
### System configuration

Hardware:
[http://developer.kaifakuai.com/product4.html](url)
Software or git version:
ESP_Easy_v2.0-20180126_normal_ESP8266_4096

TD-er, 2018-01-28:
Can you please describe on how you did flash the ESPeasy firmware?
Along with the download is also a flash tool included.
If you can set it, please also try to flash with the SPI mode set to DIO.

My Chinese is not good enough to understand your screenshot, even though it looks like "simplified Chinese" :)

yoshibatora, 2018-01-30:
I tried Release v2.0-20180130 and Release mega-20180130, both not working on my ESP-12 like board. Flashed with flash app inside the zip file, also tried using ESP download tool stil no go. But I can flash and use ESPEasy_R120_1024.bin. The flash device id shows 4026h, and 32Mbit. But I not sure it is the size of SPI flash.

TD-er, 2018-01-30:
Did you also try the _1024 version of the v2.0 or mega release?
So in short:
What firmware files did you use? (filenames)
What flash tool?
What was giving the fatal exception? Did you experience that exception with the flash tool, or was it a report of ESPeasy?
If it was a report of ESPeasy, could you give a screenshot of the error?
Can you show the flash chip information given by esptool.py? (only if you know how to run it)

yoshibatora, 2018-02-01:
I can flash with file ESPEasy_R120_1024.bin using the flash script inside the zip file below.
https://sourceforge.net/projects/espeasy/files/ESPEasy_R120.zip/download
I also tried flash ESP_Easy_v2.0-20180130_normal_ESP8266_1024.bin and ESP_Easy_v2.0-20180130_normal_ESP8266_4096.bin but end with Fatal exception error above, tried with the ESPFlashDownloadTool_v3.6.2.2 and FlashESP8266 inside the zip file.
I can use ESPEasy_R120_1024.bin, the debug info shows below.
`INIT : Warm boot #1
FS   : Mounting...
FS   : Mount successful, used 75802 bytes of 113201
INIT : Free RAM:25136
INIT : I2C
INIT : SPI not enabled
WIFI : AP Mode disabled
WIFI : AP Mode ssid will be ESP_Easy_0 with address 192.168.4.1
WIFI : Could not connect to AP!
WIFI : AP Mode enabled
WD   : Uptime 0 ConnectFailures 0 FreeMem 19240`

yoshibatora, 2018-02-01:
How to check the SPI Flash size of my ESP8266 module? When flash bins I can see 32Mbits and 4026h shows up inside the DETECTED INFO block. Does that mean I have a 32Mbit SPI Memory？

TD-er, 2018-02-01:
The debug information you show is not an indication of a failure.
It is simply stating it does not know how to connect to an accesspoint and therefore will act as one.
So you can connect to the WiFi accesspoint with SSID name "ESP_Easy_0".
Then you will be presented with a webpage where you can select a WiFi accesspoint, enter credentials and reboot.
During reboot the ESP will try to connect to the given network and start normal operations.

32 Mbits sounds like a valid amount (4 MByte)

yoshibatora, 2018-02-02:
Tried ESP_Easy_v2.0-20180130_normal_ESP8266_4096.bin, flash and boot ok. Debug info shows below.
`
INIT : Booting version: v2.0-20180130
INIT : Cold Boot
FS   : Mounting...
FS   : Mount successful, used 0 bytes of 957314
RESET: Resetting factory defaults...
RESET: Warm boot, reset count: 0
RESET: formatting...
RESET: formatting done...
FILE : Saved config.dat
FILE : Saved security.dat
RESET: Succesful, rebooting. (you might need to press the reset button if you've justed flashed the firmware)

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09826c6d
~ld
猆

INIT : Booting version: v2.0-20180130
INIT : Warm boot #1
FS   : Mounting...
FS   : Mount successful, used 75802 bytes of 957314
INIT : Free RAM:23648
INIT : I2C
INIT : SPI not enabled
WIFI : AP Mode disabled
WIFI : AP Mode ssid will be ESP_Easy_0 with address 192.168.4.1
WIFI : Could not connect to AP!
WIFI : AP Mode enabled
WD   : Uptime 0 ConnectFailures 0 FreeMem 19144
WD   : Uptime 1 ConnectFailures 0 FreeMem 15560
WD   : Uptime 1 ConnectFailures 0 FreeMem 17792
WIFI : Connecting PDCN attempt #1
WIFI : Connected! IP: 192.168.128.27 (ESP_Easy_0)
WD   : Uptime 2 ConnectFailures 0 FreeMem 17888
FILE : Saved config.dat
FILE : Saved security.dat
WIFI : AP Mode disabled
WD   : Uptime 2 ConnectFailures 0 FreeMem 20784
WD   : Uptime 3 ConnectFailures 0 FreeMem 20784
WD   : Uptime 3 ConnectFailures 0 FreeMem 20536`

TD-er, 2018-02-02:
That looks great!

So everything is working now?

yoshibatora, 2018-02-02:
Seems working OK until attached an OLED. Check this issue below.
[https://github.com/letscontrolit/ESPEasy/issues/810](url)

Grovkillen, 2018-03-01:
Can this be closed or is it still a problem?

closed 2019-10-27 by TD-er