letscontrolit/ESPEasy #781: Serial log level 4 does not show http requests

s0170071, 2018-01-28:
Just wanted to find out why the esp webserver seems to hang once in a while and found that http requests do not show up in the log.

Mega / 2.4.0

TD-er, 2018-01-28:
I noticed also the log level is a bit hard to use.
E.g. it is not possible to focus only on a certain loglevel, so if you go "deep" enough, the logs are flooded and the really interesting parts are left unnoticed.
I think the loglevel should be a bit mask. And also the selector should be more clear. So that's a bit +1 from me :)

Grovkillen, 2018-01-28:
I think the main problem is that we don't have a best practice when it comes to how we populate the log. Maybe always have the PXXX in front of all messages, and also a indication on what log level it's coming from.

[P001] [L3]: GPIO set to...

That way is always possible to filter on plugin and /or log level.

TD-er, 2018-01-28:
Yep.
@Grovkillen Did you already do a sweep on the sourcecode towards your magical "convert to wiki" interface of all the log types?

Grovkillen, 2018-01-28:
No but that's an awesome idea. Will do! :)

s0170071, 2018-01-28:
Each log entry could have a unique ID or / and  a link to the corresponding wiki page...
I need to check if the Arduino serial monitor supports clickable links...
-- 
Diese Nachricht wurde von meinem Android-Gerät mit K-9 Mail gesendet.

s0170071, 2018-01-29:
just checked... llinks are clickable if they are comments in the arduino IDE, but they cannot be clicked inside the serial monitor.

TD-er, 2018-01-29:
I think it has to be done along with #747

TD-er, 2018-01-29:
Hmm, looking into the code right now and it seems there are only 10 lines to store the logs for the web dump.
So if there are more log lines to be stored, within the refresh interval, it may not show.

Also there is a possibility for a memory leak. (write an empty string to the log) 

I will look into this.

Grovkillen, 2018-01-30:
@s0170071 please test this merged fix asap so we can close this issue. Thanks. :+1:

TD-er, 2018-01-30:
@Grovkillen As far as I know, it was not really a fix yet. If so, there was something even more horribly wrong than expected ;)
It was more of a clean-up on the Log code so we can look into things.

Grovkillen, 2018-01-30:
All right, then I know :zap:

s0170071, 2018-01-30:
So does it now show http requests?
-- 
Diese Nachricht wurde von meinem Android-Gerät mit K-9 Mail gesendet.

TD-er, 2018-01-30:
@s0170071 I tried to create serial log + web log next to eachother to show they display the same.
However, I only saw MQTT messages (on both), no HTTP messages (on neither log views)
So then I enabled Domoticz HTTP (Domoticz MQTT was also enabled).
This resulted in unstable WiFi, lots of disconnects.

So I will have a look at that also and a new feature request, to add a command to enable/disable controllers via the serial connection.

mvdbro, 2018-03-22:
Can anyone actually reproduce this bug? I've never seen serial log fail in my enviroment.

Grovkillen, 2018-04-07:
I will try to get these log pushes documented, had forgot about @TD-er s suggestion. How should we do with this issue, do a roundup on the log pushes or close and do this as a part of Knox?