letscontrolit/ESPEasy #783: Addi gibt the VEML6070 UV Sensors

sashsc, 2018-01-29:
Please add the VEML6070 Sensor to espeasy mega.

https://www.letscontrolit.com/forum/viewtopic.php?f=6&t=3085

Thanx
Sascha

luciust, 2019-03-24:
What's the status? VEM6040 (colour) is added - maybe the pure UV could be added as well?

TD-er, 2019-03-24:
Hmm, I was looking in the source, but also VEM6040 is not there.
In what plugin is it added?

luciust, 2019-03-24:
VEML 6040 was added here: https://espeasy.readthedocs.io/en/latest/Plugin/P066.html

Ii got it in the current mega_dev build. 

Actually you can get some readouts out the VEML 6070 using that plugin, but they are bogus. Gijs if you'd like to pick it up (just as with the SDS011 previoulsy) I can test it: got the setup up an running. Would be great to have real UV sensor. 

https://learn.adafruit.com/adafruit-veml6070-uv-light-sensor-breakout/arduino

TD-er, 2019-03-24:
I have not written the VEML 6040 support.
I also don't have both VEML sensors, but I can see why having a UV sensor may be useful.  For example, I would like to have one of these near my Anycubic Photon printer, since the resin should not be exposed to UV light unless meant to.
For testing/debugging a sensor, I really should have a sensor myself, or else it will take way too much time.

luciust, 2019-04-02:
Just noticed that ESPEasy Playground has the VEML6070 plugin (https://github.com/letscontrolit/ESPEasyPluginPlayground/blob/master/_P213_VEML6070) by @jowiemann

Compiled and seems that is working fine, tested with lamp, and UV lamp. More testing is needed:

[image]

TD-er, 2019-04-03:
So his issue is now more about moving it to the main repo?
These sensors are on their way from Ali Express to my mailbox.

luciust, 2019-04-03:
Yes, seems we have support - only the sensor needs some testing - I've got UV lamp, also the quartz glass, regular glass, will update within two weeks.

RemCom, 2019-04-09:
I have a few esp's with a VEML 6040 on it. I'm currently in the process of switching from Tasmota to ESPeasy. I'm currently halfway porting the VEML 6040 code from there to here. The new module is a little further developed than the code shared up here. I think i got the time to finish and test it by end this week.

RemCom, 2019-04-09:
Had some time this evening so i finished it already. Made a merge request for it: #2440

TD-er, 2019-04-10:
@RemCom I made a few review suggestions on the PR.
They are not that big.
Maybe if there's not much difference between the VEML 6040 and the 6070 they can be done in the same plugin?

luciust, 2019-04-10:
I thought so, the 6070 is read in the same way. Merging both hw version in one plugin seems plausible.

Anyway - waiting for dev- versions I've got one running and testing with classic glass as well as quartz glass for outdoor.

RemCom, 2019-04-10:
@TD-er I will have a look and resolve the comments tonight. I didnt make de i2c address configurable because it always uses address 0x38 and 0x39 but i can make it configurable.

@luciust The 6070 is a little different and using an 8 bit registers instead of 16 bit in case of the 6040

We sure can merge them into 1 plugin if we want to but they both need their own reads and write (Or make a 8/16 switch on the function). @TD-er Let me know if you prefer it to be merged with the 6040.

TD-er, 2019-04-10:
I'm not sure yet.
They seem to have a lot of code overlap, but they do measure different things.
I will think about it.

luciust, 2019-04-10:
@RemCom This is IMHO - But I believe that there will be cases when someone would like to use them both  - since one is defacto color and the other is UV sensor - so the good compromise (dev job vs. usability) would be to keep'em apart.

RemCom, 2019-04-10:
For start i will resolve the change requests so that they are clean.

RemCom, 2019-04-10:
@TD-er Just pushed the updates. I already set most Changes requested to resolved (this is the procedure from where i work). If you instead wanted to do this sorry for already doing it.

luciust, 2019-05-06:
@TD-er - do we have that in dev release? Haven't seen in the changelog, just pinging if we could get in in main dev release

TD-er, 2019-05-07:
@luciust I merged the PR for this plugin.
Should be included in the next nightly build

luciust, 2019-05-07:
@TD-er Thanks, will do testing when it shows up!

closed 2019-10-10 by TD-er

ventilator, 2020-08-02:
I have this VEML6070 UV sensor connected but do not find it selectable in the most recent built mega-20200721. Was the plugin really merged?

TD-er, 2020-08-02:
It is plugin 084, which is in the "testing" build as can be seen [here](https://espeasy.readthedocs.io/en/latest/Plugin/_Plugin.html)

Egyras, 2020-09-08:
> It is plugin 084, which is in the "testing" build as can be seen [here](https://espeasy.readthedocs.io/en/latest/Plugin/_Plugin.html)

strange but for me VEML6070 it works with Tasmota, but not with ESPEasy.....

TD-er, 2020-09-08:
What error message do you see in the logs?
And maybe it would be useful then to open an issue about it, instead of keeping discussing below a closed thread?