letscontrolit/ESPEasy #784: Sysinfo freezes the ESP8266 (Wemos D1)

micropet, 2018-01-29:
If I integrate the plugin "Generic - System Info" and set WiFi RSSI, the ESP stops responding.

Greetings Peter

TD-er, 2018-01-29:
Can you give a bit more information?
Like build version and perhaps also some screenshots of the settings?
I have a similar setup here, with the Wemos D1, and WiFi RSSI and it is working fine.

Make sure you don't set the Delay value too low. (default = 60)

micropet, 2018-01-29:
[image]
[image]
[image]

Hi,

Delay ist 60. Software is always the newest Trunk from Github, Core 2.4.0

My Rules are:

// ----- ESP-207 Ki
On System#Boot do    //When the ESP boots, do
	Publish %sysname%/IP,%ip%
	timerSet,1,60      //Set Timer 1 for the next event in 60 seconds
	timerSet,2,20
endon

On Rules#Timer=1 do  //When Timer1 expires, do
	Publish %sysname%/IP,%ip%
	Publish %sysname%/Time,%systime%
	Publish %sysname%/Uptime,%uptime%
	timerSet,1,60       //Resets the Timer 1 for another 60 seconds
endon
// -----

on Motion#Pir>0 do
	If %systime% > 08:00:00  // nur tagsueber anzeigen
		If %systime% < 20:00:00 	
			gpio,14,1  // Led2 pink ON
		endif	
	endif	
endon

on Motion#Pir=0 do
     gpio,14,0  // Led2 pink OFF
endon
// -----

On MH-Z19#PPM<700 do
    NeoPixelAll,0,50,0,0  // gruen
endon
	 
On MH-Z19#PPM>700 do
	if [MH-Z19#PPM]<1200
		NeoPixelAll,50,50,0,0 // gelb
	else 
		NeoPixelAll,50,0,0,0 // rot
	endif 
endon	
// -----

TD-er, 2018-01-29:
We haven't tested with 2.4.0 core lib.
Can you also check with a build using 2.3.0?
See definitions in PlatformIO.ini with platform set to:
`platform = espressif8266@1.5.0`

Adding %rssi% isn't such a bad idea, since it can be used in a lot of other use cases and using a Task/Plugin for it is maybe a bit overkill.

micropet, 2018-01-29:
Gijs, i use Arduino and all my other Projects are Core 2.40.
I do not want to go backwards.

I hope you can install the %rssi% parameter.  :)

Peter

TD-er, 2018-01-29:
Can you check it?
It is only one line of code as can be seen in the PR.

micropet, 2018-01-29:
Yes - it woks!
Many thanks.

20:12:01 ESP-212/status Connection Lost
20:12:01 ESP-212/status Connected
20:12:02 ESP-212/IP 192.168.0.238
20:13:02 ESP-212/Time 20:13:01
20:13:02 ESP-212/Uptime 1
20:13:02 ESP-212/RSSI -69

closed 2018-01-30 by Grovkillen

TD-er, 2018-01-30:
Just as a note if someone ever stumbles upon this.
I have still no idea why the ESP froze  when adding the SystemInfo plugin.