letscontrolit/ESPEasy #789: GlobalSync fail when having long variable names

Grovkillen, 2018-01-30:
https://www.letscontrolit.com/forum/viewtopic.php?f=18&p=23953&sid=4e551714542a884be52c101922f1e7c3#p23953

> i did some testing with GlobalSync, which did not work wen using long device-variable-name on the sending node, 
renamed that device on the local device name and it started working, was able to display the outside temp on livingroom-esp.
that is working, nice feature !

Grovkillen, 2018-01-31:
Bumped this to 2.1.0 since it's a fairly big update that is needed (not only to fix this problem but a general cleanup and better structure is needed).

psy0rz, 2018-02-02:
indeed :) for now disable globalsync if you have weird crashes or stability issues. (or if you dont need it)

uzi18, 2018-02-14:
@Grovkillen @psy0rz valuenames and tasknames have fixed lenght (max 25 chars)

closed 2019-10-11 by Grovkillen