letscontrolit/ESPEasy #790: Please support logical operators like AND and OR in rules

micropet, 2018-01-30:
At the moment it is very inconvenient to make a rule that is only active at a certain time.

I would like to dim or increase the brightness of my lighting within certain times.

e.g.

If% systime%> 23:50:00 OR if% systime% <01:00:00

Grovkillen, 2018-01-30:
We'll most likely not add that functionality to the current rules engine. But you can do what you want using existing rules.

On Clock#Time=All,01:00:00 do
 Event,EventName
EndOn

On Clock#Time=All,23:50:00 do
 Event,EventName
EndOn

On EventName do
 //do what ever
EndOn

micropet, 2018-01-30:
Thanks for the hint. I will try it. 
What is "uPy Easy"?

TD-er, 2018-01-30:
That's the micro-Python based version mainly developed by LisaM.
It is only just into Beta phase, so still in full development.
But already it is clear that version will be lot easier to maintain and add extra features.

Grovkillen, 2018-01-30:
It's the future firmware architecture with will aim future ESP32 and multiple other hardware MCU units.

micropet, 2018-01-30:
[image]

Wau, just flasht on an ESP32 Devboard. Look very good.

Peter

uzi18, 2018-01-30:
Is there uPy for esp8266?

[image: upeasy]
<https://user-images.githubusercontent.com/2838584/35587566-77a31c58-05fe-11e8-9a8c-47448904f4eb.jpg>

micropet, 2018-01-30:
I have it not seen.
I think you must compile yourself.

uzi18, 2018-01-30:
It is on micropython.org/download ....

TD-er, 2018-03-02:
Is this already supported via PR #932 ?

uzi18, 2018-03-02:
We can close it

closed 2018-03-02 by micropet

Grovkillen, 2018-03-13:
https://www.letscontrolit.com/wiki/index.php/Tutorial_Rules#AND.2FOR

uzi18, 2018-03-13:
Lots of new stuff now :)