letscontrolit/ESPEasy #791: Add new command to enable/disable controllers

TD-er, 2018-01-30:
See the commands list: https://www.letscontrolit.com/wiki/index.php/ESPEasy_Command_Reference
There should be some command to enable/disable controllers and tasks.

Grovkillen, 2018-03-15:
Should go for devices/tasks as well?

Grovkillen, 2018-10-23:
This is essentially a variant of the `taskclear` and `taskclearall` commands. So `taskdisable` and `taskdisableall` would be the new commands. And thus the controller related and notification related commands would be similarly named.

uzi18, 2018-10-23:
maybe to. temporary disable, and save command to save this new configuration?

TD-er, 2018-10-23:
During crash recovery the tasks are "temporary" disabled, which is working fine. (it is disabled in settings, not saved)
So we could add these for sure.

qwertangel, 2019-11-09:
+1. it would be very useful to enable/disable tasks from rules. I need to start SR04 distance measure when my garage door opens. and with enable/disable it would be possible.

tonhuisman, 2023-09-03:
This has been resolved, as the commands are implemented. Can be closed,

closed 2023-09-03 by TD-er