letscontrolit/ESPEasy #792: Browser blocked

s0170071, 2018-01-30:
240/lwip1/mega:
If I access the ESP with Linux/Firefox, it works fine at first. If I let it alone for a few minutes, I cannot access the ESP any more. 
Is the ESP crashed ? No- because I can still access it using Android.
Is the Browser faulty ? No- because if I reset the ESP, I can access it again without doing anything to the browser.

TD-er, 2018-01-30:
Can you test running a ping to the ESP?
Just to check there is indeed a WiFi connection.

s0170071, 2018-01-30:
there is... because I can still access it using Android.

s0170071, 2018-01-30:
ping does not go through...

TD-er, 2018-01-30:
Hmm we may have to look into these issues, but also start the discussion on what core libs we will support.
There are just too many issues and too little active developers at the moment.
And some of these "core lib related" issues really take a lot of time.
For example the chunked issue I had already took the same amount of time I usually need to fix about 5+ other issues.

Not saying that you're on your own, definitely not, but more that we really have to decide on importance now given the very limited dev time we have.

s0170071, 2018-01-30:
I'll try 230 then to see if its a real issue.

s0170071, 2018-01-30:
ok, flashed 240 again and now its still running. Maybe some variable which is finally initialized now. I will cut the power and try again tomorrow. 
Puh, that is just the kind of issue that I love. Like a bug from a psycho movie. Only I can see it- look again- its gone. And when you least expect it....
As for debugging: I do what I can but it seems like I have a lot less time and experience than you. 
Btw, how is your chunking bug comning?

TD-er, 2018-01-30:
The chunking bug is now parked, since I felt I was not going anywhere with it.
About these kind of bugs, I was planning to force a full settings erase of all data elements of that plugin when you add a new plugin.
Just to make sure no left over settings may cause issues.

I guess the controller issue I created today may be something similar.

s0170071, 2018-01-30:
For a moment I thought I got the bug. I was getting a program CRC fail message. 
Again and again....
on other boards too.... 
With different cables...
different IDEs... 

Until I found that someone changed the CRC code
from   
`             calcBuffer[buf] = pgm_read_dword((void *)i+buf*4);                                         // read 4 bytes
`
to 
`               calcBuffer[buf] = pgm_read_dword((uint32_t*)i+buf*4);                                         // read 4 bytes
 `
I have to admit, that at first glance that looks OK, but in fact it isn't- pointer arithmetic is the key word. Changing from void* to uint32* and adding sth to that pointer gives pointers to different memory locations.

`               calcBuffer[buf] = pgm_read_dword((uint32_t*)i+buf);                                         // read 4 bytes
 `
would be correct.
@TD-er, would you be so kind and correct that. I am currently not on good terms with Gitkraken.

TD-er, 2018-01-30:
Ah, indeed, my bad.
I will correct it later this evening.

TD-er, 2018-01-30:
I was thinking, maybe I should change it to 
`calcBuffer[buf] = pgm_read_dword((uint8_t *)i+buf*4); // read 4 bytes`

Or else it may be you have to force 4-byte alignment?

The reason why I changed it was because of compiler errors, since the compiler doesn't allow pointer arithmetic on void pointers.

s0170071, 2018-01-30:
I guess the alignment is provided by the linker. I just use the addresses and lengths as I find them in the binary.
The start address is always aligned as it never changes. If the length is ever not multiple of four either version of the code will fail as we read dwords.
So... Your pick ;-)

https://github.com/letscontrolit/ESPEasy/issues/792#issuecomment-361722731

s0170071, 2018-01-31:
cant reproduce it any more... maybe a hardware hickup.

closed 2018-01-31 by s0170071