letscontrolit/ESPEasy #793: PWM (fade) command is blocking the loop.

Grovkillen, 2018-01-30:
https://www.letscontrolit.com/forum/viewtopic.php?f=6&t=4740

This should not happen since a fade is most likely not on sub second level.

psy0rz, 2018-02-02:
do we have a pwm fade?? (cant find it) if so: doing a fade requires many updates per seconds, since you want to fade to be smooth. for this reason its usually blocking.

Grovkillen, 2018-02-02:
Yes we have (not documented on the wiki though). It's the third value after PWM... if left empty no fade is done.

https://github.com/letscontrolit/ESPEasy/blob/9cbaa4d7e7345a7e7912d146452848be4265a0bf/src/_P001_Switch.ino#L231

Grovkillen, 2018-02-02:
Maybe we need a new command called PWM_ramp or something like that? I don't want to break any old stuff.

zdzichu, 2018-02-03:
It is documented on wiki: https://www.letscontrolit.com/wiki/index.php/GPIO#Commands :
`PWM,<GPIO>,<state>,<duration>`
_duration_ is for fading.

Grovkillen, 2018-02-03:
Thanks @zdzichu , it's myself who added it on 13 January 2018‎ .... go figure! :laughing:

tonhuisman, 2023-09-03:
This seems to be solved, so can be closed.

closed 2023-09-04 by TD-er