letscontrolit/ESPEasy #797: Add KNX-IP Protocoll to ESPEasy

sisamiwe, 2018-01-30:
I'm wondering whether it would be possible to add the KNX-IP protocoll to ESPEasy. There is already a good basis [here](https://github.com/envy/esp-knx-ip).
This would provide the ESPEasy possibilities to a broad range of useres.

There is the need to configure the group adresses. See this [postings](https://knx-user-forum.de/forum/%C3%B6f ... lan-an-knx) as example.

TD-er, 2018-01-30:
I had to google it to have a vague idea of what it should do.
Is it somewhat similar to MODBUS? Like reading/writing registers from multiple devices on the bus.
Or more like M-bus? Query a device-address and get all sensor values at once.

And another -more practical- problem is how do we test it as developer?
I am not going to develop/integrate something, build a test version, send it to someone to test it, get some vague description of an error, etc.
Such development will take like forever and I don't have hairs enough left on my head for those development strategies any more ;)
What I meant to say is that those devices look quite expensive for just testing.
I am not really sure ESPeasy will be the first option people will choose when working with such devices.

sisamiwe, 2018-01-31:
I understand your concerns, in terms of development and the needed infrastrukture for testing. A simple way would be a RPI running knxd as KNX interface. So there would be just 2 bus members the ESP and the knxd running on the RPI. knxd is also in [github](https://github.com/knxd/knxd)
So from my opinion there is no need for "real" KNX devices for the test setup.

In terms of application. KNX is used for professional applications, but there is t trend do add "nice things" like Sensores or LED-Dimmers to KNX on a cheaper way. There are already some applications developed like SONOFF (WLAN switch based on ESP) to interact with KNX-IP.
Here are some sketches for [that](https://knx-user-forum.de/forum/%C3%B6ffentlicher-bereich/knx-eib-forum/1158719-esp8266-%C3%BCber-wlan-an-knx). It's in german

Whydo you think, that ESPEasy is not the right way?

TD-er, 2018-01-31:
ESPeasy is not a 'realtime' application. Sure it can do some time critical things like communicate to a sensor. But often the ESP is then somewhat like a master. It will send some signals to a sensor and the sensor will reply with some values, or start a measurement.
It is also possible to listen to some GPIO pins with an interrupt callback coupled to it. SoftwareSerial does that to allow some decent speeds on serial communications and not miss a bit (we hope).
And even if it misses a bit, it is often only a single measurement and CRC will no longer match so not much lost.

But listening to a communications bus where missed messages will be annoying to say the least is not really what ESPeasy was meant to do.
Worst-case scenario is that some kind of critical on/off command will be missed, with results ranging from annoying to disaster.
I am not familiar with KNX, but given some of the issues reported at the project you linked, gave me some concerns about it.

So that's a bit of my concerns.
For coupling with a profi system, it simply has to work very stable, or else people will not adopt it.

If we can just start very simple, by adding sensors which just have to send some values every now and then, it could just work. But a quick glance at the 'specsheet' of the Gira products, they were talking about encrypting the communications. Encryption is a good thing, but often these systems user some proprietary layer which can be very hard to maintain compatible with upcoming releases of the proprietary systems.
And given the limited number of active programmers (about 10-15 hours per week) I think it will take a lot of resources for probably not a big group of users.

If you can prove me wrong, please do, since these concerns are based on about 15 minutes browsing of me.

henfri, 2018-02-01:
The communication to Knx would be via IP. So there should be Buffers, No?

KNX is very Common in buildings in Europe

Grovkillen, 2018-02-01:
But should the plugin target a specific type of KNX gateway or what? I haven't found anything about being able to communicate directly using HTTP or other open IP protocol.

https://www.weinzierl.de/index.php/en/all-knx/knx-devices-en/produktarchiv-en/knx-ip-interface-730-en

TD-er, 2018-02-01:
@henfri If it is TCP/IP or UDP it would indeed be kept in buffers.

As I said before, I have 0 experience with KNX and my main concern is the development/maintenance time (and costs) involved vs. the number of users/usecases.

psy0rz, 2018-02-02:
isn't there some software project that implements a MQTT to KNX bridge?

Grovkillen, 2018-02-02:
Here's an python library which talks directly to a KNX/IP tunnelling server. But then again, you need that gateway. And are they all using the same communication?

The OP need to dig more into this and help us out if something will be considered for ESP Easy.

henfri, 2018-02-02:
Hello,

there are existing solutions to communicate from ESP Easy to KNX. These work via MQTT or generic UDP. For this, smarthome-ng can be used. See [here ](https://knx-user-forum.de/forum/supportforen/smarthome-py/1189828-einbindung-esp-mit-espeasy-firmware)and [here](https://github.com/smarthomeNG).

But:
The a key concept in KNX is a decentralized structure. Thus, many users would prefer a solution without a central server.

This is the reason for this feature request and I think this is valid.

KNX/IP is a standard and the solution from https://github.com/envy/esp-knx-ip implements this for the ESP. This is also actively in use by some users. Thus, I see no technical showstopper.

With ESP Easy I see a very userfriendly solution that would really be a breakthrough for this usecase. I think that there is a huge possible user basis.

Greetings,
Hendrik

henfri, 2018-02-03:
Hello,

what do you think about this feature-request? 
If you are bought in, what can I do to support?

Regards,
Hendrik

TD-er, 2018-02-03:
I would love it to add new features that will enable a bigger audience for ESPeasy.

But my biggest concern is that combining with a more professional grade system also puts some expectations on usability. And I am afraid it may take a lot of development time to get there.

I will take a look at it, but I guess getting v2.0 ready should have a higher priority now at this moment.
For future development, we should make some roadmap or at least make an overview of the direction ESPeasy should take.
Currently there is a number of ESP firmware versions (Tasmota, ESPeasy, Espurna to name a few) and I guess we should try to prevent doing all the same.
ESPeasy is more into allowing to combine sensors to integrate into other systems.
In that sense, it makes sense to add support for KNX-IP. But to decide on that, I first want to know how it works, to get an idea on how the integration should be done.

There is now also a parallel development of uPyEasy, which currently is getting up to speed in development. That's based on Python and currently more aimed at a bit more powerful devices like the ESP32.
We also have to decide on how these projects will work towards eachother.

And there is still the problem that testing often only can be done using real hardware, which is rather expensive for KNX.

henfri, 2018-02-03:
Hello Gijs,

thanks for your reply. I understand your concern about the diverse landscape of ESP firmeware versions. I know of none that supports KNX though.
I can support on what is needed and how the integration should look like from the user point of view. Also it is possible to set up a Docker Container or a Raspi image that can be used for testing with no hardware cost. I am not so familiar with the market so that I do not know how many devices would support the ESP32.

Greetings,
Hendrik

ascillato, 2018-04-06:
Hi,

I would love to see KNX IP protocol support on ESPEasy!

A very good KNX IP Library for ESP8266 is [envy/ESP-KNX-IP](https://github.com/envy/esp-knx-ip)

KNX IP Protocol uses a UDP multicast on 224.0.23.12:3671, allowing to any device to communicate to each other without the need of any server or router. KNX is a decentralized system.

There is no need to buy any KNX devices for using the KNX IP protocol.

Besides, I use KNX on [Home Assistant](https://github.com/home-assistant/home-assistant).

HomeAssistant has a [XKNX Python Library](https://github.com/XKNX/xknx) to connect to KNX devices using a KNX Router. As I don't have a **KNX Router**, I use the software [KNXd](https://github.com/knxd/knxd) on the same Raspberry Pi than Home Assistant.

I use knxd as a Software **KNX Router** just for reading this UDP Multicast. It has other features that need extra hardware like connect to KNX devices by Twister Pair, Power Line or RF.

Currently, I'm working on a [MOD](https://github.com/ascillato/Sonoff-Tasmota_KNX) to add KNX support for [Sonoff-Tasmota](https://github.com/arendst/Sonoff-Tasmota) using [ESP-KNX-IP Library](https://github.com/envy/esp-knx-ip). Hope to have a working version in few days. After that, If you want, I can help to add KNX to ESPEasy

henfri, 2018-04-06:
Good news.
Thanks!

giggo, 2018-08-17:
Is this being worked on actively?

TD-er, 2018-08-17:
@giggo Not that I am aware of.

ascillato, 2018-08-18:
Hi,

Up to now, I haven't had the time to work on a KNX Driver for ESPEasy. Sorry.

The KNX Driver for Tasmota is working really good and it is now part of the official Tasmota. I'm using it a lot. With this, now it is very easy to have a Sonoff to Sonoff direct communication without any other device or server in between (only your WIFI Router). You can turn on/off a sonoff from another sonoff or send sensor values, etc. Also it is added an option for enhanced KNX communication to prevent any telegram loss due that KNX IP Protocol uses Multicast and some Wifi Routers are not so reliable with that. The communication with a Real KNX Network has been proved to be very good. Also, now Home Assistant has everything needed for KNX, so with the precompiled version called Hassio, you can use KNX enabled devices at once. For more information please see https://github.com/ascillato/Sonoff-Tasmota_KNX

So, if there is anyone interested on start working on a KNX Driver for ESPEasy, you can take a look to the driver file https://github.com/arendst/Sonoff-Tasmota/blob/development/sonoff/xdrv_11_knx.ino

The libraries required are:
* https://github.com/arendst/Sonoff-Tasmota/tree/development/lib/esp-knx-ip-0.5.1 (both envy versions merged in one)
* https://github.com/arendst/Sonoff-Tasmota/tree/development/lib/ESPAsyncUDP-master (already patched)

**Libraries explanation**

This driver uses the ESP-KNX-IP Library from [envy](https://github.com/envy/esp-knx-ip). Envy in his repository has 2 versions of this library. One uses esp core lib 2.3.0 and up and the other 2.4.1 and up. The main difference is that the 2.4.1+ is a little faster but it adds [another library](https://github.com/me-no-dev/ESPAsyncUDP) also. This other library also needs a patch.

I have a modified ESP-KNX-IP Library at https://github.com/ascillato/esp-knx-ip that have a on-compilation-time option to choose between 2.3.0+ and 2.4.1+ versions (Tasmota is using this Library). Also you can find these libraries at:
* https://github.com/arendst/Sonoff-Tasmota/tree/development/lib/esp-knx-ip-0.5.1 (both envy versions merged in one)
* https://github.com/arendst/Sonoff-Tasmota/tree/development/lib/ESPAsyncUDP-master (already patched)

Hope this helps. Please, do not hesitate on asking anything you need.

TD-er, 2018-08-18:
@ascillato That sounds very good.
As soon as a new ESPeasy release is made, I really must have a look at this.

henfri, 2019-06-09:
Hello,

(a neutral questions without trying to be pushy):
Has any progress been made on this?

Regards,
Hendrik