letscontrolit/ESPEasy #799: Wemos OLED 64x48 inverted SSD1306 Framed issue.

mattlward, 2018-01-31:
I am running on Wemos hardware D1 Mini 3.0.0 and Wemos 64x48 OLED display.  

I am running Release v2.0-20180130 dev.  

If I configure the display for normal orientation, I get the name of the unit in small font and my single line in large font.   Works great!

If I switch to rotated with the following config the screen displays my 1 line of large font and a bunch of dots across the bottom of the screen.
[image]

Bad output:
[image]

I have reset, loaded other versions and no changes.  It also happens on multiple units.

TD-er, 2018-01-31:
I guess some parts are a bit too hard-coded based on 64 pixel height.
Those dots are page tabs showing the active page. They appear at the bottom.

Is the text on the top line readable?
Can you also show a photo of the normal orientation? Then I can see if all objects are present, or maybe the display does some scaling

mattlward, 2018-01-31:
The text on the top line never appears.  I can take pics this evening when I get home.  All objects are present when the display is flipped.  The dots are not on the screen when it is not rotated.

mattlward, 2018-02-01:
Pic of the unit with normal orientation.  The screen looks great here, just upside down when mounted.
[image]

TD-er, 2018-02-01:
Is it an idea to include flags to display header text and footer tabs?
And to define a number of lines offset for the text field?

mattlward, 2018-02-01:
Not sure what you are asking... I can get the data to the display and sized correctly without rotated display.  But if one wall mounts one of these the USB needs to come out of the bottom of the unit and requires a rotated display to look right and keep the cord right.

TD-er, 2018-02-01:
The OLED framed plugin does make a lot of assumptions about the dimensions of the display.
Those are fine for 128x64 pixel displays, but not for other sizes.
So I could allow to remove the upper and/or lower indicators and allow for the "text area" to be shifted a few pixels to allow to center the important data.
Your display is 64x48 and there are also 128x32 displays.

mattlward, 2018-02-01:
The Wemos 64x48 display is perfect for tiny sensor stacks.  I for one will 7 or 8 of them when done and would love it if the display looked the same inverted as normal.  It is nice to see the name of the device for setup reasons.

mattlward, 2018-02-06:
Loaded the latest and now it has 2 page indicators at the bottom when inverted, that is the number of pages I have in use.  It did have one for each slot before.  Also the contrast function works great on the little display!

TD-er, 2018-02-06:
Indeed, it is meant to be showing only a dot for each page containing information.

I've updated quite a lot on that plugin and also made some preparations for different resolutions.
The ones I want to support -in the end- are:

- 128x64
- 128x32
- 64x48

The display library should support them by now. (also SH1106 based OLED displays are now supported in v2.0)
But I still have to try to re-arrange the plugin to support the smaller displays.

mattlward, 2018-02-06:
I appreciate the work on the 64x48.  It would be nice to have the name of
the system and the 1 line of data like in normal when inverted.

emeyedeejay, 2018-11-01:
Love the framed plugin - just wondering when the 64x48 (Wemos OLED shield) support for it might be available?

Re the system name etc. this should be optional - sometimes it's great, sometimes, not - should be configurable.

PR #3050 (2020-07-23): [P036] Memory optimization (issue #2799) and other improvements
CHG: Memory optimization (issue #2799)
CHG: Added setting for 'Disable scrolling while WiFi is not connected',
         scrolling is not smooth as long as the ESP tries to connect Wifi (Load 100%!)
BUG: Wifi-Bars not fully right for 128 pixel wide displays
BUG: clearing window for scrolling lines was too high and to narrow
BUG: content of lines with pixel width > 255 had to be shortened
CHG: scrolling lines improved
CHG: variables renamed for better understanding

Fixes: #2312
Closes: #3026 
Fixes: #3127 
Fixes: #2760 
Fixes: #279 [... cut]
closed 2020-07-23 by TD-er