letscontrolit/ESPEasy #801: Support for WPA2-Enterprise [feature request]

Haocen, 2018-01-31:
**NOTE: This is not a support forum! For questions and support go here: https://www.letscontrolit.com/forum/viewforum.php?f=1**

### Steps to reproduce
How can we trigger this problem?

Cannot connect to WPA2-enterprise PEAP

Does the problem presist after powering off and on? (just resetting isnt enough sometimes)

### Expected behavior
Tell us what should happen?

Should be able to since SDK had implement this around a year ago.

### Actual behavior
Tell us what happens instead?

### System configuration

Hardware: WEMOS D1 mini

Software or git version:

v2.0

I looked at this project: 
https://github.com/martin-ger/esp_peap_psk

Seemed doable, I can contribute to to the ui part of the config if we plan to do it.

TD-er, 2018-01-31:
What version of the Arduino ESP8266 core libraries are needed for this?
For the v2.0 branch of ESPeasy we will stick to lib version 2.3.0
For the Mega branch of ESPeasy we are testing with the 2.4.0 core lib, but that still has a lot of issues. (although it is much faster on the web interface, so would be really nice to have it.)

When you build it yourself, there is a build option in PlatformIO.ini to build against the 2.4.0 libs.
Make sure the latest updates are installed in PlatformIO (and active => restart PIO)
Then you can also test if it is already working with the latest core lib.

N.B. the 2.4.0 core lib is not yet supported by us due to limited dev time available.

Haocen, 2018-01-31:
The project I linked indicated it is supported since SDK v2.1 , I'll try it out myself later today to see how it goes.

Haocen, 2018-02-01:
Sorry a misunderstanding on my side, I guess when saying 2.4.0 core lib you're referring to https://github.com/esp8266/Arduino/releases/tag/2.4.0 which contains the v2.1 SDK the v2.3 core lib contain v1.5.3 SDK and does not contain WPA2-ENTERPRISE support. I'll wait for it became stable then.

tonhuisman, 2023-09-03:
This seems to be solved, so can be closed.

TD-er, 2023-09-04:
Nope, it has not been solved as the WPA2-Enterprise needs certificates.