letscontrolit/ESPEasy #804: OLED cannot display the degree symbol.  " ° "  Don't know if it should or not.

Budman1758, 2018-02-01:
### Steps to reproduce
How can we trigger this problem?
Attempt to display a degree symbol on either a Framed or a regular OLED display

Does the problem presist after powering off and on? (just resetting isnt enough sometimes)
Yes

### Expected behavior
Tell us what should happen?
Symbol should show in the display

### Actual behavior
Tell us what happens instead?
On the Framed display the symbol does not show.  On the regular OLED the entire line that should have the symbol is garbage.
The symbol shows and saves correctly in the web UI.  Using the the method to type the symbol from this link.  [https://www.google.com/search?q=type+degree+symbol&oq=type+degree+symbol&aqs=chrome..69i57.12952j0j4&sourceid=chrome&ie=UTF-8](url)
[image]
[image]

The degree symbol should be showing just ahead of the "F" after the temp number.  Like the title says, don't know if it's supposed to work or not.

Grovkillen, 2018-02-01:
As far as I understand these characters need to be hard coded into the plugin in order to work. My Swedish ÅÄÖåäö is not part of that hexadecimal list either. See here:

https://github.com/letscontrolit/ESPEasy/blob/cae3baf71e5de0872112b366df7cceae8f50be39/src/_P023_OLED.ino#L336

So I suggest you change the topic to "Add degree character to OLED plugin" or something :)

HV-NL, 2018-02-01:
I had the same problem, so I tried every possible ASCII character. In this way I found that 0x7F is the character for degrees Celcius

This is the code I use in Domoticz to send a display string to the OLED display, attached to ESPEasy:
```
-----------------------------------------------------------------------
-- display_statusregel
-----------------------------------------------------------------------
function display_statusregel(regelnr, sLine)
    runcommand = "curl -m 1 'http://192.168.11.165/control?cmd=OLED," .. regelnr .. ",1," .. sLine .. "'"
    os.execute(runcommand)
end -- display_statusregel

    -- stuur de huidige CV temperatuur naar het OLED display op de schoorsteenmantel
    sLine = "CV%20In%20%20:%20" .. sCV_In .. "%20%7FC%20%20"
    display_statusregel(1, sLine)

```
[image]

v-a-d-e-r, 2018-02-01:
Copy this to get the degree sign --> Â°
All without changing the source code....

Budman1758, 2018-02-01:
> Copy this into the text field (before F or C) to get the degree sign --> Â°

That does not seem to work for me......

> This is the code I use in Domoticz to send a display string to the OLED display, attached to ESPEasy

Not using a controller.

TD-er, 2018-02-01:
I could add a replacement check to change for example "{D}" into the character needed for the degree sign on the screen?

Budman1758, 2018-02-01:
Whatever works works for me. This is not a big deal.  I know you guys are pretty busy chasing real bugs and more important stuff.

TD-er, 2018-02-01:
Well, given the number of issues on the OLED-Framed plugin, I guess it is one of the most frequently used plugins.

And I also would like the degree sign to show on my own devices ;)

psy0rz, 2018-02-02:
Also we should fix the bug that shows garbage for unknown characters. (instead show nothing or a questionmark or something)

Grovkillen, 2018-02-02:
And add box character (filled and non-filled) to be used as progress bars.

TD-er, 2018-02-02:
@Grovkillen that is another thing. The included font does not have bars.

Let's put bars in another issue to show charts.

TD-er, 2018-02-03:
It should be fixed when @psy0rz does his magic merging :)
Also added lots of other symbols.

Try these (short versions)  (Euro does not work yet)
```
{D} {<<} {>>} {u} 
{E} {Y} {P} {c} 
{^1} {^2} {^3} 
{1_4} {1_2} {3_4}
{+-} {x} {..} 
```

You can also use the same characters via cut 'n paste from Character map.
And HTML versions of these (like `&deg;` or `&micro;` ) which are probably easier to use in other ways of communications (JSON/HTML/MQTT)

Still have to add command support for the OLED Framed plugin.

Grovkillen, 2018-02-04:
To all : the list is found here
https://www.letscontrolit.com/wiki/index.php/OLEDDisplay#Special_characters

These character codes can be used not only for the OLED but also other places as well.

Budman1758, 2018-02-08:
Halfway there.  Works great in Framed OLED. (degree symbol)
Reg OLED still has garbage.
mega-20180206

TD-er, 2018-02-08:
The regular OLED plugin does use the parseTemplate function:
https://github.com/letscontrolit/ESPEasy/blob/eb2dbb5b90da52ffb6aa460cf0eab5f59eef107d/src/_P023_OLED.ino#L187-L203

But it uses an old interface to embedded fonts and these extra symbols are not included in the font.

closed 2018-02-10 by Grovkillen