letscontrolit/ESPEasy #810: Daily flash write rate exceeded error

yoshibatora, 2018-02-02:
**NOTE: This is not a support forum! For questions and support go here: https://www.letscontrolit.com/forum/viewforum.php?f=1**

### Steps to reproduce
How can we trigger this problem?
Using SSD1306 OLED and modify devices settings several times. FS : Daily flash write rate exceeded! (powercycle to reset this) shows.
Does the problem presist after powering off and on? (just resetting isnt enough sometimes)
Reboot helps.
### Expected behavior
Tell us what should happen?
I cannot modify OLED settings and an error message like this shows.
[image]

`Log
187 : INIT : Free RAM:23648
187 : INIT : I2C
187 : INIT : SPI not enabled
1070 : WIFI : AP Mode disabled
1070 : WIFI : AP Mode ssid will be Mario_0 with address 192.168.4.1
1070 : WIFI : Connecting PDCN attempt #1
5227 : Current Time Zone: DST time start: 2018-03-25 02:00:00 offset: 60 minSTD time start: 2018-10-28 03:00:00 offset: 0 min
5230 : WIFI : Connected! IP: 192.168.128.27 (Mario_0)
6599 : WD : Uptime 0 ConnectFailures 0 FreeMem 21296
6638 : DHT : Temperature: 6.00
6638 : DHT : Humidity: 46.00
36599 : WD : Uptime 1 ConnectFailures 0 FreeMem 21000
37421 : DHT : Temperature: 6.00
37422 : DHT : Humidity: 46.00
38632 : FS : Daily flash write rate exceeded! (powercycle to reset this)
38633 : FS : Daily flash write rate exceeded! (powercycle to reset this)
38639 : FS : Daily flash write rate exceeded! (powercycle to reset this)
66659 : WD : Uptime 1 ConnectFailures 0 FreeMem 21128
67697 : DHT : Temperature: 6.00`
### Actual behavior
Tell us what happens instead?

### System configuration

Hardware:
ESP-12E type ESP8266 module made by KAIFAKUAI. More info about the modules shows below.
[https://github.com/letscontrolit/ESPEasy/issues/url](url)
Software or git version:
ESP_Easy_v2.0-20180130_normal_ESP8266_4096.bin

Grovkillen, 2018-02-02:
This happens for a reason. You can either reboot the unit or use this command to reset the count to zero.

`resetFlashWriteCounter`

closed 2018-02-02 by Grovkillen

Grovkillen, 2018-02-02:
As a side note: please use the forum first. Only report here if you are for sure that you have found a genuine bug.

TD-er, 2018-02-02:
And as a little explanation on why this happens.
The flash memory has a limited number of writes possible before it wears out and becomes unusable.
To protect the flash, ESPeasy only allows up to 100 writes per 24h.
When it happens and you want to continue writing to it, please follow the steps @Grovkillen advised.